trezor/trezor-firmware #640: [New App] NEM2 Support

ScaredIbis, 2019-10-22:
This issue proposes support for the [next iteration of the NEM blockchain](https://nemtech.github.io/guides/migration/migrating-from-nis1-to-catapult.html), Known as Catapult. This will introduce a new app (nem2) so as to retain the functionality of the existing nem integration.

the nem2 app will likely be similar in size to the existing integration but with a new set of transaction types and a different signature scheme, both of which are incompatible with the existing nem implementation. 

The [NEM2 desktop wallet](https://github.com/nemfoundation/nem2-desktop-wallet) will be the supporting application for this integration. There is a small team that has been working on Trezor support in the wallet app.

The following are links to relevant sections in the developer documentation:
- [getting started](https://nemtech.github.io/getting-started/what-is-nem.html)
- [key pairs and addresses](https://nemtech.github.io/concepts/cryptography.html#cryptography)
- [transaction types](https://nemtech.github.io/concepts/transaction.html#transaction-types)
- [signing transactions](https://nemtech.github.io/concepts/transaction.html#signing-a-transaction)

A transaction explorer app can be found [here](https://github.com/nemfoundation/nem2-explorer)

prusnak, 2019-10-22:
What will happen to NEM1 once NEM2 is launched?

ScaredIbis, 2019-10-22:
NEM1 will continue to exist once NEM2 is launched, they will be seperate chains.

Details about the XEM ticker are to-be-determined, but NEM2 will be a new coin.

prusnak, 2019-10-22:
OK, feel free to work on the code, but we will merge it only if NEM2 is in TOP30 coins at coinmarketcap.

closed 2019-10-22 by prusnak

ScaredIbis, 2019-10-24:
@prusnak I noticed after diving into the docs and examples of past PRs from other projects, it seems that people are using mostly python to write their apps. We have a [project](https://github.com/nemtech/catbuffer) used to generate c++ code for serialising our different transaction types that we were planning on using for this implementation. I just wanted to make sure it's not expected that the serialisations are written in python so that I don't waste my time building it this way. Do you have any thoughts on this?

matejcik, 2019-10-24:
We will not be adding C++ code to the codebase, generated or otherwise. You will need to write the serialization in Python

matejcik, 2019-10-24:
@ScaredIbis is a description of the serialization format available?

evias, 2019-10-24:
Hello @matejcik, thank you for coming back to us so quickly :+1:

NEM2 uses a project called `catbuffer` to handle serialization across all client applications (and SDKs). This tool is inspired from `flatbuffer` tool specialized for data serialization compatible with Catapult protocol.

Following are the highlights / schema with regards to transaction serialization:

- Custom types: https://github.com/nemtech/catbuffer/blob/master/schemas/types.cats
- "Entities": https://github.com/nemtech/catbuffer/blob/master/schemas/entity.cats
- Base transaction data: https://github.com/nemtech/catbuffer/blob/master/schemas/transaction.cats
- Example of *specialized serialization schema for Transfer Transactions*: https://github.com/nemtech/catbuffer/blob/master/schemas/transfer/transfer.cats

Hope this answers your question?

Also, noted on the Python serialization requirement.

matejcik, 2019-10-24:
> Hope this answers your question?

It does not, unfortunately. I'm interested in seeing the serialization encoding - akin to [protobuf encoding description](https://developers.google.com/protocol-buffers/docs/encoding)

evias, 2019-10-24:
ok sorry,

an example of serialization with NEM2 (for a transfer transaction) is as follows: https://github.com/nemtech/nem2-sdk-typescript-javascript/blob/master/src/infrastructure/catbuffer/TransactionBuilder.ts#L167

evias, 2019-10-31:
Hey there @matejcik and @prusnak, 

does only **serialization** need to be done with Python? Or does this rule count for the signing procedure as well?

We were thinking about a C implementation because the legacy NEM integration was done with C.

Thank you

matejcik, 2019-11-01:
> We were thinking about a C implementation because the legacy NEM integration was done with C.

That is because the legacy codebase is all in C. The new trezor-core codebase is in Python, except for hardware-related handling and the cryptography library (which is a separate subproject). This is to ensure maximum run-time safety. So yes, all code must be Python; have a look at `core/src/apps/nem` to see how it works for NEM1.

this is of course only if you want your code included in the official firmware, which is not happening until NEM2 is in Top30 anyway.

evias, 2019-11-05:
Thank you. 

I wrongly assumed the codebase state and I am thankful for this information. I will be circling back in case there is a problem with our integration.