trezor/trezor-firmware #641: PKCS#11 library to use instead of custom ssh agent (also usable with other PKCS#11-supporting applications)

hiviah, 2019-10-22:
It would be nice to have a PKCS#11 library which can be used in `PKCS11Provider` in SSH, also in many other applications that support PKCS11 (e.g. Firefox/Thunderbird, client certificate login, S/MIME, etc).

PKCS#11 requires some set of functions to be defined in the standard. Here is a short overview - https://docs.oracle.com/cd/E19120-01/open.solaris/819-2145/chapter2-11/index.html

IIRC it is not necessary to implement each one, just the authentication/signature/digest.

prusnak, 2019-10-22:
Does PKCS#11 mandate physical connection (e.g. it has to be CCID via USB) or are we free to use anything we want?

hiviah, 2019-10-22:
I think it's up to whatever you want to use. PKCS#11 is just a shared library that defines some functions, but does require any physical characteristics (like CCID).

hiviah, 2019-11-01:
SSH just added direct support for U2F and FIDO2, so we won't need PKCS#11 or custom agent anymore - https://marc.info/?l=openssh-unix-dev&m=157259802529972&w=2.

closed 2019-11-01 by hiviah