trezor/trezor-firmware #643: Send ButtonRequestTypes automatically using ui.Layout

tsusanka, 2019-10-23:
We could send ButtonRequests automatically in ui.Layout when the screen changes. Specifics to be discussed.

matejcik, 2019-10-24:
General idea: there is a "logical structure" for each layout, which could be represented by a protobuf type and a field in `ButtonRequest`, and the actual on-screen layout would be derived from this structure. This would also pave the way for layouts that work on both TT and T1.

Problems:
* if the logical structure is generated from trezor-core code (perhaps named after the layout class name?), changes to the code could be breaking for users of `ButtonRequest`
* related: if we wanted to auto-generate protobufs from code, there doesn't seem to be a good way to ensure consistent numbering of the fields

perhaps one would first need to add the protobuf definition, and generate code scaffold from that?

matejcik, 2021-09-07:
superseded by layouts refactor

closed 2021-09-07 by matejcik