trezor/trezor-firmware #647: "Firmware is too big" issue

SheldonYS, 2019-10-24:
I try to compile current `master` branch code from GitHub and update firmware to Trezor One. However, when I upload firmware.bin to Trezor with `make upload` command, I got "Firmware is too big". How to solved this issue? Does Trezor One support firmware v2.1.8?

OS platform: Mac OS Mojave 10.14.6
arm-none-eabi-gcc version: 8.3.1

SheldonYS, 2019-10-24:
One more question, does T1 firmwares stop upgrading? Because I did not find the code for v1.x.y

SheldonYS, 2019-10-24:
Yes. I find it. I'll try this. Thanks a lot!

SheldonYS, 2019-10-24:
When I build `master` branch for T1, I got `bootloader has to be smaller than 32736 bytes (current size is 32832)` error. But it is fine for building tag `legacy/v1.8.3`. Do you have the same problem with `master` branch?

prusnak, 2019-10-24:
Yes, bootloader in `master` fails at the moment. Use `1.8.3` for now.

Big bootloader issue reported here: https://github.com/trezor/trezor-firmware/issues/649

bosomt, 2019-12-14:
works fine with latest 1.8.4 build