trezor/trezor-firmware #653: Error installing dependencies from pipfile.lock

mdhz, 2019-10-29:
I ran the following command which gave the error attached at the end:

sudo ./build-docker.sh
I also tried sudo -H ./build-docker.sh

Please help me with the error. I need the legacy bootloader for my custom made Trezor. I don't mind using a pre-compiles bootloader if that's available directly from the official trezor team.

The error comes after python environment starts.

The error is:
`Installing initially failed dependencies…
[pipenv.exceptions.InstallError]:   File "/usr/local/lib/python3.7/dist-packages/pipenv/core.py", line 1874, in do_install
[pipenv.exceptions.InstallError]:       keep_outdated=keep_outdated
[pipenv.exceptions.InstallError]:   File "/usr/local/lib/python3.7/dist-packages/pipenv/core.py", line 1253, in do_init
[pipenv.exceptions.InstallError]:       pypi_mirror=pypi_mirror,
[pipenv.exceptions.InstallError]:   File "/usr/local/lib/python3.7/dist-packages/pipenv/core.py", line 859, in do_install_dependencies
[pipenv.exceptions.InstallError]:       retry_list, procs, failed_deps_queue, requirements_dir, **install_kwargs
[pipenv.exceptions.InstallError]:   File "/usr/local/lib/python3.7/dist-packages/pipenv/core.py", line 763, in batch_install
[pipenv.exceptions.InstallError]:       _cleanup_procs(procs, not blocking, failed_deps_queue, retry=retry)
[pipenv.exceptions.InstallError]:   File "/usr/local/lib/python3.7/dist-packages/pipenv/core.py", line 681, in _cleanup_procs
[pipenv.exceptions.InstallError]:       raise exceptions.InstallError(c.dep.name, extra=err_lines)
[pipenv.exceptions.InstallError]: ['Collecting cryptography==2.8', '  Downloading https://files.pythonhosted.org/packages/be/60/da377e1bed002716fb2d5d1d1cab720f298cb33ecff7bf7adea72788e4e4/cryptography-2.8.tar.gz (504kB)', '  Installing build dependencies: started', "  Installing build dependencies: finished with status 'done'", '  Getting requirements to build wheel: started', "  Getting requirements to build wheel: finished with status 'done'", '    Preparing wheel metadata: started', "    Preparing wheel metadata: finished with status 'done'", 'Building wheels for collected packages: cryptography', 'Failed to build cryptography']
[pipenv.exceptions.InstallError]: ["WARNING: The directory '/.cache/pipenv/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.", "WARNING: The directory '/.cache/pipenv' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.", "  WARNING: Building wheel for cryptography failed: [Errno 13] Permission denied: '/.cache'", 'ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly']
ERROR: ERROR: Package installation failed...
`

prusnak, 2019-10-29:
Btw, latest bootloader binary is available from https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/bootloader.dat

mdhz, 2019-10-29:
> Running `./build-docker.sh` works for me. There is something wrong with your configuration.

I am running docker in 32 bit Ubuntu using https://gitlab.com/docker-32bit/ubuntu

Could it be that?

mdhz, 2019-10-29:
> Btw, latest bootloader binary is available from https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/bootloader.dat

Thank you.

mdhz, 2019-10-29:
> Btw, latest bootloader binary is available from https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/bootloader.dat

How do I convert this into binary file?
Can you help me please? Renaming it to .bin is showing some weird ASCII characters and I assumed that it's the wrong way.

prusnak, 2019-10-29:
```
wget https://raw.githubusercontent.com/trezor/trezor-firmware/master/legacy/firmware/bootloader.dat
mv bootloader.dat bootloader.bin
```

prusnak, 2019-10-29:
> I am running docker in 32 bit 

most probably

mdhz, 2019-10-29:
> ```
> wget https://raw.githubusercontent.com/trezor/trezor-firmware/master/legacy/firmware/bootloader.dat
> mv bootloader.dat bootloader.bin
> ```

Thanks. I will try now.