trezor/trezor-firmware #656: Electrum cannot pair with your Trezor

bosomt, 2019-10-29:
Electrum cannot pair with your Trezor error.
latest 2.1.8 regular firmware with seeds loaded.
But Electrum works fine. You see both devices connected + you are able to check QR,address, XPUB.
Send of transaction wasn't tested.
Nixos , Electrum 3.3.8

[image]
[image]

log from console: https://paste.ofcode.org/mFk5mPKK8jB6eXgHRxiDMU

bosomt, 2019-10-29:
+ both devices show 2 of 2 + same address :)

[image]

bosomt, 2019-10-30:
tested with older firmware and this error goes way back at least up to 2.1.0 firmware

bosomt, 2019-10-30:
same result with 2.0.8

bosomt, 2019-10-30:
same results with Windows 7 + Electrum

matejcik, 2019-10-30:
please add steps to reproduce

bosomt, 2019-10-30:
[image]
[image]
[image]
[image]
[image]
[image]

prusnak, 2019-11-01:
Oddly enough this does not happen when one of the devices is T1. In other words, both devices need to TT.

prusnak, 2019-11-01:
The error comes from here: https://github.com/spesmilo/electrum/blob/ccccaf099faa9d3dc920d6895f4e0c4b58fd8e5f/electrum/plugin.py#L496

matejcik, 2019-11-15:
@bosomt can you reproduce the problem if you make sure you *only* have two Trezors connected?

matejcik, 2019-11-15:
i am 87 % sure this is an issue on Electrum side, so closing in favor of https://github.com/spesmilo/electrum/issues/5759

closed 2019-11-15 by matejcik