trezor/trezor-firmware #666: dry-run recovery allows setting u2f counter, label, passphrase protection

matejcik, 2019-10-31:
This is an artifact of the fact that all these things are parameters of the `RecoveryDevice` message. There is an explicit guard against changing PIN, but not the other settings.

Not sure how T1 behaves.

I don't think this is a security problem, but it's something that should not happen IMO. I propose to change logic in core so that if `msg.dry_run` is set, all other parameters of the message are ignored.

prusnak, 2019-11-10:
Please also include device test which tests this.

matejcik, 2019-11-18:
There are differences between behavior of core and legacy.

core explicitly rejects `pin_protection` and silently sets label, u2f counter and passphrase protection.

legacy silently ignores all fields that do not pertain to dry-run recovery.

If we want to make the protection stronger, we'll need to implement it in legacy too. I'm still of the opinion that silently ignoring the fields here is an acceptable behavior though.

bosomt, 2019-12-14:
@matejcik 

is it possible that you broke it somehow ?
i tried latest master 2.1.9 a started dry run.
It doesn't even start. Passphrase is not enabled + it's fresh backup-ed device without PIN or passphrase enabled

Action was interrupted Error details: Forbidden field set in dry-run: passphrase_protection

[image]

bosomt, 2019-12-14:
with 1.8.4 it says: Action was interrupted Error details: Forbidden field set in dry-run

[image]

bosomt, 2019-12-14:
log: ["log","Sat, 14 Dec 2019 07:10:21 GMT","[trezor.js] [call] Sending","Initialize",{}]
["log","Sat, 14 Dec 2019 07:10:21 GMT","[trezor.js] [call] Received","Features",{"vendor":"trezor.io","major_version":1,"minor_version":8,"patch_version":4,"bootloader_mode":null,"device_id":"3E1F859ED081F7845BCCBD5B","pin_protection":false,"passphrase_protection":false,"language":null,"label":null,"initialized":true,"revision":"d6cc035f009feedcae413b733221dff053e6b6c5","bootloader_hash":"f7fa165be6d780f3e1af00abc07df8b3076bcdad72d70da22a63d8896b6391d8","imported":null,"pin_cached":false,"passphrase_cached":false,"firmware_present":null,"needs_backup":false,"flags":null,"model":"1","fw_major":null,"fw_minor":null,"fw_patch":null,"fw_vendor":null,"fw_vendor_keys":null,"unfinished_backup":false,"no_backup":false,"recovery_mode":null,"capabilities":[1,2,5,7,8,10,12,14],"backup_type":null,"sd_card_present":null,"sd_protection":null,"wipe_code_protection":false}]
["log","Sat, 14 Dec 2019 07:10:21 GMT","[trezor.js] [call] Sending","RecoveryDevice",{"word_count":24,"type":0,"dry_run":true,"passphrase_protection":false,"enforce_wordlist":true}]
["log","Sat, 14 Dec 2019 07:10:22 GMT","[trezor.js] [call] Received","Failure",{"code":"Failure_DataError","message":"Forbidden field set in dry-run"}]

bosomt, 2019-12-14:
it's broken in FW and not in wallet.
same results in trezorctl 

trezorctl recovery-device -d 
Traceback (most recent call last):
  File "/nix/store/1pz9gbmf373fkpisdq2paix84n1xmfkh-python3.7-trezor-0.11.5/bin/.trezorctl-wrapped", line 9, in <module>
    sys.exit(cli())
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/trezorlib/cli/trezorctl.py", line 537, in recovery_device
    dry_run=dry_run,
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/trezorlib/device.py", line 148, in recover
    u2f_counter=u2f_counter,
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/trezorlib/tools.py", line 233, in wrapped_f
    return f(client, *args, **kwargs)
  File "/nix/store/yik42axy7z43rn9w32vw1dxfnkk2qcld-python3-3.7.5-env/lib/python3.7/site-packages/trezorlib/client.py", line 222, in call
    raise exceptions.TrezorFailure(resp)
trezorlib.exceptions.TrezorFailure: DataError: Forbidden field set in dry-run

prusnak, 2019-12-15:
@bosomt Actually it's broken both in web wallet and in trezorctl :-)

What you see in the web wallet is not yet fixed issue https://github.com/satoshilabs/mytrezor/issues/873 

About trezorctl failure: @matejcik was this fixed recently there and boso needs a newer version or what's causing the problem?

matejcik, 2019-12-16:
yes, the issue in trezorctl was fixed [in the same commit](https://github.com/trezor/trezor-firmware/commit/feccf53afabacc05464cc027b5538a375975442f#diff-fd24f080824f50de4214c04117357c07L133-L145) as the rest. This is currently unreleased, so we'll also need to push a new version of trezorlib before the firmware goes out.

prusnak, 2019-12-16:
OK, closing then. Let's not forget to release web wallet and trezorctl fixes :)

closed 2019-12-16 by prusnak

bosomt, 2019-12-22:
fixed in latest 49 stage wallet

bosomt, 2019-12-29:
@matejcik @prusnak @TomasZorvan 
FW release in 3 days, please don't forget to release wallet and trezorctl fix

matejcik, 2019-12-30:
trezorctl 0.11.6 which implements this is out

admoseley, 2020-01-26:
Is this fixed. I tried to test my seed phrase and got this message. My wallet is on the latest firmware.

prusnak, 2020-01-26:
@admoseley which FW version do you use?

commit 34913a32 (2019-11-20): all: disallow most RecoveryDevice fields in dry-run (fixes #666)