trezor/trezor-firmware #671: Rethink derivation paths caching

tsusanka, 2019-11-01:
We have a C cache for derivation paths implemented in trezor crypto's [bip32.c](https://github.com/trezor/trezor-firmware/blob/master/crypto/bip32.c#L509-L586).

We also have some caching in

tsusanka, 2019-11-01:
Results of bip32 speed test can be found in https://github.com/trezor/trezor-firmware/issues/663 .

tsusanka, 2019-11-04:
Let's also have a look on Keychain caching. Currently, on each message we create a new instance of Keychain IMHO.

tsusanka, 2020-04-28:
> Let's also have a look on Keychain caching. Currently, on each message we create a new instance of Keychain IMHO.

Gist of a brief discussion with @matejcik:
- We indeed currently create a new Keychain for every new message.
- Because of that the root cache is not maintained between the messages.
- We are not sure want/need to modify that. So we are leaving it as it is at the moment.

matejcik, 2020-04-29:
The trezor-crypto cache works rather different from what I was imagining.

There is a static slot for **one** root HDNode, and 10 cache slots for derivations from that root.

In current core code, we set up a namespace which internally derives and caches a root node.
For example, the namespace is `m/44'/714'`.
When querying `m/44'/714'/0'/0/0`, first the node `44'/714'` is derived from root `m`, and the result node is stored in Keychain.
On the C level, root `m` is cached with one derivation item `44'/714'`.

Subsequently, from the Keychain node, suffix `0'/0/0` is derived.
On C level, root `m` is replaced by root `m/44'/714'`, and one item added to its cache `0'/0/0`.
If the Python code now requests node `m/44'/714'/0'/0/1`, the Keychain root `m/44'/714'` is reused. This matches what is stored in the C cache, so a second item is inserted for suffix `0'/0/1`.

Keychain does not cache the leaf nodes, so when requesting `m/44'/714'/0'/0/0` again, it attempts to re-derive suffix `0'/0/0`. This is already cached in C, so we don't actually re-derive anything and the result is returned immediately.

BUT when the workflow is restarted and a new Keychain instance is created, it again requests derivation of `m/44'/714'`, which replaces the C-cached root with `m` and the previous cache entries are lost.

---

This is made even more fun by Bitcoin, which actually registers namespace "", so the cached root is always `m`, and all derived paths are cached until the user switches currencies.

(As it happens, deriving 0 or 1 levels does _not_ use the cache. Requesting re-derivation of `m` will return a deterministic result, and getting any suffix from it will use the cache, because the cached root is still `m` from previous executions. This is what bit us in the UI tests: the cache survives a wipe, and by a series of happy coincidences still contains "valid" values when repeatedly asking for a Bitcoin Testnet address).

---

I am actually not sure what to do about this. Disabling the C cache without Python-level replacement _will_ degrade performance. Perhaps not by much, but we will need to evaluate this.

On the Python level, we can design a smarter cache that works with namespaces and can cache items for multiple roots. This is of course a complication of the existing code and we will need to evaluate how badly we actually need it.

matejcik, 2020-07-14:
bumping priority as this now relates to #1038 

With the C-level caching disabled as of 7d5b26d02e18d99133779ce6b84ef31a2b843b29, there is currently no derivation path cache. This might matter when signing big transactions with repeated paths (i.e., when consolidating 100 inputs sent to the same address).

The usecase is very limited, but still might be worth investigating.