trezor/trezor-firmware #672: FR: Ability to generate a Shamir Backup for an existing private key (or a 24-word seed)

EDmitry, 2019-11-02:
I wonder if this is technically possible at all. The scenario I envision: 

- memorize 24 word sequence
- have Shamir backups for the same private key as a backup for cases if the user forgets the seed (although there is always a question about how to remember who/where Shamir backups were distributed to in cases of severe memory loss).

This would be especially useful if a 24 word sequence was already memorized and a Shamir back-up was decided to be used later.

EDmitry, 2019-11-02:
Looks like it's technically possible but will generate 59 words per share for BIP39 mnemonics: https://github.com/trezor/python-shamir-mnemonic/issues/15

That's slightly problematic, but better than nothing, considering that there are still only 24 words to memorize (for 256bits of entropy), and Shamir seeds can be stored physically.

The other question, going in reverse, if a Shamir Backup generated, is it possible to create some sort of memorizable seed that will singlehandedly generate the private key?

matejcik, 2019-11-04:
It is possible, yes, but the resulting length is 59 words, which is lousy for usability. That is also the reason we don't offer this option by default.

It would be possible to make a converter firmware where you enter your 24-word seed and get back a set of 59-word shares, but this feature probably won't ever make it into the official firmware.

Re memorability, that's easy with Super Shamir (available in beta now). Create a 1-of-1 group with one 20-word share, which you memorize - and, say, 3-of-5 group with five 20-word shares, which you store in five secure locatons and use for back-up.

Of course, as we keep repeating, if you insist on memorizing *something*, make a long and memorable passphrase instead. Less constraints, more flexibilty, better memorability.

prusnak, 2019-11-04:
I think everything has been answered.

closed 2019-11-04 by prusnak

EDmitry, 2019-11-04:
Thank you for an elaborate answer, @matejcik! In hypothetical situation if one of the 5 shares gets compromised, would it be possible to regenerate that group while keeping the 1-of-1 group intact, with the same 20-word seed?

matejcik, 2019-11-05:
> Thank you for an elaborate answer, @matejcik! In hypothetical situation if one of 5 shares gets compromised, would it be possible to regenerate that group while keeping the 1-of-1 group intact, with the same 20-word seed?

That would be pointless, as you can't invalidate the old group. Even if you generate a new one (which can be done), the old one can still be used to recover the secret.

And of course, you cannot generate a new secret to be recovered with the old 1-of-1 group -- because that group already recovers the _old_ secret.

matejcik, 2019-11-05:
Hmm but you could destroy the remaining 4 shares and generate a new 3-of-5 group. That would work, yes. If you encode the same secret again, you will get a different set of shares.

jonathancross, 2020-04-08:
> If you encode the same secret again, you will get a different set of shares.

@matejcik Will the Trezor Model T let you encode the same secret a second time?

matejcik, 2020-04-09:
> > Create a 1-of-1 group with one 20-word share
> 
> @matejcik Is this possible? Current [Super Shamir documentation](https://wiki.trezor.io/User_manual:Creating_a_wallet_with_Super_Shamir) says that the minimum number is 2.

yes, because Super Shamir with one group is just Regular Shamir ;)
there's two groups in what I proposed: the 1-of-1, and a 3-of-5 or something.

> 
> > If you encode the same secret again, you will get a different set of shares.
> 
> Will the Trezor Model T let you encode the same secret a second time?

No.
It would not be difficult to make a modified firmware for this exact purpose though: take the official FW, remove the check that ensures backup is run only once. Install that on your device, recover from the original seed, do a backup. It should be fully secure (in theory), because you're using "almost-official" firmware, the only modifications you did yourself so you can trust that, and the seed is only ever entered on the Trezor device.

sportvikings1981, 2021-05-29:
i’m trying to claim free flare spark tokens. I had my xRP on my trezor t at time of snap shot. I have 2 20 seed Shamir backup. If I can figure out how to create single 24 backup (from Shamir), I can link my trezor to a spark compatible wallet so flare can see Xrp was there at time of snapshot. Is this possible?