trezor/trezor-firmware #673: Feature Request: BIP39 checksum generation

fresheneesz, 2019-11-03:
I would like to use seed words generated off device (eg using something like [this](https://www.stavros.io/posts/perfectly-secure-bitcoin-wallet-generation/) or [this](https://github.com/taelfrinn/Bip39-diceware)), however the checksum isn't feasible to generate without the help of a computer, and generating on a different computer would compromise the seed. 

I propose that the trezor allows inputting 23 seed words for a new wallet, and then generates the 24th word for the user. This way the user can generate external entropy without compromising it. This allows for slightly less trust that must be put in Trezor and their associated manufacturing pipeline.

tsusanka, 2019-11-05:
That'd be nice I guess, but in general we prefer users to generate the seed inside Trezor, because otherwise uneducated users could enter seeds from books, their favourite songs etc, which is discouraged. I'm leaving this as an idea though, thank you.

tsusanka, 2020-08-04:
Hi there, thank you for your suggestion. After a thorough discussion in our Product team we have decided we do not wish to implement this feature. We need to prioritize our backlog and focus on features that bring the most value to our customers. We also need to restrict the feature set to keep the firmware small and maintainable. Thank you for your understanding.

closed 2020-08-04 by tsusanka