trezor/trezor-firmware #674: Latest 2.1.8 regular freezing problem

bosomt, 2019-11-03:
Latest 2.1.8 regular freezing problem

Since i loaded white noise background i'm experiencing freezing of device.
It's not GUI problem, device doesn't communicate with computer.
It's not happening every time but often. and always during address confirmation or sending of transaction. 

See video.
I will try to remove white noise background and i Will continue troubleshooting.

https://photos.app.goo.gl/dpK2vN1KEjxY2auQA

bosomt, 2019-11-03:
Even it's not responding wallet/computer see it and you see error after you physically disconnect it

bosomt, 2019-11-03:
And yes it' mem related.
Problem is gone after i load it with standard background.
And it's back after I load white noise background

tsusanka, 2019-11-04:
Could you try on 2.1.7 and 2.1.6?

bosomt, 2019-11-04:
i'm on it

bosomt, 2019-11-04:
it's broken in 2.1.5 :(
same behavior

bosomt, 2019-11-04:
it's broken in 2.1.6
same behavior

bosomt, 2019-11-04:
2.1.4 has different behavior, but freezes too.
only backlight goes off, but device can be waken up

prusnak, 2019-11-09:
Closed in favor of https://github.com/trezor/trezor-firmware/issues/684

closed 2019-11-09 by prusnak

bosomt, 2019-12-14:
works fine with latest 2.1.9 with WN background
I will keep using WN background and if anything goes wrong I will reopen or create new issue