trezor/trezor-firmware #677: rework USB interfaces for debug firmware

prusnak, 2019-11-05:
As can be seen in [usb.py](https://github.com/trezor/trezor-firmware/blob/master/core/src/usb.py) we currently disable `webauthn` USB interface for debug firmware so we'll have endpoints for `debug` AND `vcp` interfaces. We can rework this to have both `debug` AND `webauthn` interfaces for the debug build (so only `vcp` will be disabled).

Current:

firmware regular: `webauthn`
firmware bitcoinonly: -
firmware regular debug: `debug` `vcp`
firmware bitcoinonly debug: `debug` `vcp`

New:

firmware regular: `webauthn`
firmware bitcoinonly: -
firmware regular debug: `debug` `webauthn`
firmware bitcoinonly debug: `debug` `vcp`

Emulator already has all interfaces enabled for debug build (`debug`, `vcp`, `webauthn`), so we don't need to change the behaviour there.

matejcik, 2019-11-05:
But we need the `vcp` interface to see the console output, right? It should be configurable which of `webauthn`, `vcp` will be available in the debug fw

prusnak, 2019-11-05:
I described the minimum requirements. Having this easily configured would be a plus +1

prusnak, 2019-11-09:
WIP in [`reorder-usb`](https://github.com/trezor/trezor-firmware/tree/reorder-usb) branch - no configuration yet.

prusnak, 2019-11-13:
PR in https://github.com/trezor/trezor-firmware/pull/701

closed 2019-11-13 by prusnak