trezor/trezor-firmware #680: trezorctl throws when trying to sign EOS delegatebw action

EDmitry, 2019-11-08:
% cleos -u https://api.eosnewyork.io convert unpack_action_data eosio delegatebw <data hash>
{
  "from": "<account>",
  "receiver": "<account>",
  "stake_net_quantity": "0.2000 EOS",
  "stake_cpu_quantity": "2.0000 EOS",
  "transfer": 1
}

% trezorctl eos-sign-transaction -n "m/44'/194'/0'/0/0" --file transaction.json
Traceback (most recent call last):
  File "/usr/local/bin/trezorctl", line 10, in <module>
    sys.exit(cli())
  File "/Library/Python/3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Library/Python/3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/3.7/site-packages/click/decorators.py", line 27, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/Library/Python/3.7/site-packages/trezorlib/cli/trezorctl.py", line 1542, in eos_sign_transaction
    return eos.sign_tx(client, address_n, tx_json["transaction"], tx_json["chain_id"])
  File "/Library/Python/3.7/site-packages/trezorlib/tools.py", line 233, in wrapped_f
    return f(client, *args, **kwargs)
  File "/Library/Python/3.7/site-packages/trezorlib/eos.py", line 325, in sign_tx
    header, actions = parse_transaction_json(transaction)
  File "/Library/Python/3.7/site-packages/trezorlib/eos.py", line 307, in parse_transaction_json
    actions = [parse_action(a) for a in transaction["actions"]]
  File "/Library/Python/3.7/site-packages/trezorlib/eos.py", line 307, in <listcomp>
    actions = [parse_action(a) for a in transaction["actions"]]
  File "/Library/Python/3.7/site-packages/trezorlib/eos.py", line 270, in parse_action
    tx_action.delegate = parse_delegate(data)
  File "/Library/Python/3.7/site-packages/trezorlib/eos.py", line 147, in parse_delegate
    sender=name_to_number(data["sender"]),
KeyError: 'sender'

It looks like Cleos tool uses "from" field name, but trezorctl expects "sender".

prusnak, 2019-11-09:
@EDmitry can you try to fix the bug and send us a pull request?

prusnak, 2019-11-09:
Do you know where are these fields are documented so we can check every one of them? I would say there is some kind of API documentation somewhere for EOS.

EDmitry, 2019-11-10:
Yes, the ABI is generated based on this file:

https://github.com/EOSIO/eosio.contracts/blob/56483d05cb02b7fb263fef6013c3780ce114fd4e/contracts/eosio.system/include/eosio.system/eosio.system.hpp#L556

ZdenekSL, 2019-11-26:
Fixed

closed 2019-11-26 by ZdenekSL