trezor/trezor-firmware #682: Investigate random ordering of FIDO2 Device tests

prusnak, 2019-11-09:
Follow-up of https://github.com/trezor/trezor-firmware/issues/449

We now have FIDO2 Device Tests, but they fail unless we use the particular random-order-seed: https://github.com/trezor/trezor-firmware/commit/b530e7c1053f6ddcd9929ec1f71ba25a0d586eaf#diff-7fcd8f4a5caeaad75fdeaca6b937a07dR35

We think this might be caused by resident crendetials, but needs more investigation.

andrewkozlik, 2019-11-19:
I think I fixed the FIDO2 tests to work in random order (https://github.com/solokeys/fido2-tests/commit/ac88a8cf09f962ba397cc6030492b71c27440e88) and I also increased the response timeout to avoid random failures in CI (https://github.com/solokeys/fido2-tests/commit/67f0023be4913a86f62bedf700f86ca3079cf842). There is one unresolved issue, in some cases pytest is quicker than the emulator, so while the workflow from the previous command is being cleaned up, the next test in the test suite issues a command and receives a BUSY response. This happens for example with `--random-order-seed=704546`. I tried to fix this issue by closing the workflow before sending the response to a command but that broke the FIDO2 error dialogs (https://github.com/trezor/trezor-firmware/compare/andrewkozlik/682). So we need make the test suite wait for the device to clean up after the previous test, for example by adding something like `sleep(0.1)` after each test.

matejcik, 2019-11-19:
@andrewkozlik is it possible to query the device through FIDO repeatedly until it stops being busy?
I would propose injecting such snippet (through a pytest hook) in front of every test. This could even not be Trezor-specific, but it can be if we need that.

andrewkozlik, 2019-11-21:
I solved the busy issue by allowing any outstanding FIDO2 workflows to be terminated and replaced if another command is received after a response to the previous command was issued, e.g. the next command terminates any error popups from the previous command workflow. PR: https://github.com/trezor/trezor-firmware/pull/718

I ran the tests about 37 times and they failed two or three times due to responses taking too long, e.g. a ping response which should have been returned within 100ms took about 340 ms. The emulator debug log showed that in these cases the command itself was processed quickly. The problem was that it took about 300ms for the processing to start. We can look into this further, but for now it should be sufficient to allow the tests to be rerun.

PR #718 (2019-11-26): Enable random ordering of FIDO2 device tests
Resolves #682
closed 2019-11-26 by matejcik