trezor/trezor-firmware #692: Check and document reproducible builds

tsusanka, 2019-11-10:
Part of #685.

Check or add a guide how to build firmware yourself and check it against the provided by us. There is some content on https://wiki.trezor.io/Developers_guide-Deterministic_firmware_build but it is outdated.

prusnak, 2020-06-07:
I just checked the wiki and it seems pretty fine - i.e. there is no outdated stuff there.

tsusanka, 2020-06-12:
I have tried this for 2.3.1 and the fingerprints did not match. So this is most likely caused due to the fact that we have introduced NixOS after 2.3.1. So let me try again during the next release and let's close then.

mmilata, 2021-11-15:
I've reviewed https://wiki.trezor.io/Developers_guide:Deterministic_firmware_build and it seems correct, at least for recent versions.

* `legacy/v1.9.0` + `core/v2.3.0`: build gets stuck due to micropython submodule that fell off github
* `legacy/v1.9.4` + `core/v2.3.5`: first version with newer micropython, builds fine, comparison to official fw OK
* `legacy/v1.10.3` + `core/v2.4.2`: builds fine, comparison OK

Verifying legacy FW is a bit complicated due to the legacy header. #1918 attempts to improve the description of the process.

closed 2021-11-16 by mmilata

bosomt, 2021-11-17:
Does it needs QA or i can move it to approved issues section ?

tsusanka, 2021-11-17:
Let's approve, Martin basically tested himself :).

PR #1918 (2021-11-16): docs(common): add section on reproducible builds
Fixes #692, fixes #1728. Related: #1713. The _deterministic build_ page is based on https://wiki.trezor.io/Developers_guide:Deterministic_firmware_build with slightly different instructions for T1.