trezor/trezor-firmware #693: Check and document hardware

tsusanka, 2019-11-10:
Part of #685.

Let's check the current state of the hardware section and fill in the blanks.

tsusanka, 2019-11-10:
Also see https://mcudev.github.io/trezor-model-t-dev-kit/index.html for a great write-up.

mcudev, 2021-01-16:
i was thinking about adding to `trezor-firmware/docs/hardware/model-t/boot.md` a description about where the development signing and public keys come from (to show that they are not magic -- obviously they are not, but it takes some searching through the code to figure it out). this demonstrates how the keys are generated. something like the following:

`pip3 install PyNaCl`
`python3.8`
```
# for reference:
#     https://ed25519.cr.yp.to/
#     https://crypto.stackexchange.com/a/68129
#     trezor-firmware/core/embed/boardloader/main.c
#     trezor-firmware/core/embed/bootloader/main.c
#     trezor-firmware/python/src/trezorlib/firmware.py
#     trezor-firmware/python/src/trezorlib/_internal/firmware_headers.py
#     trezor-firmware/core/embed/vendorheader/vendor_unsafe.json
import nacl.signing
signing_key = nacl.signing.SigningKey(bytes.fromhex('41' * 32))
print(bytes(signing_key).hex()) # '4141414141414141414141414141414141414141414141414141414141414141'
print(bytes(signing_key.verify_key).hex()) # 'db995fe25169d141cab9bbba92baa01f9f2e1ece7df4cb2ac05190f37fcc1f9d'
print(bytes(signing_key.to_curve25519_private_key()).hex()) # 'a8800471de52f74e8d323237b1b913ca28bc0f46f26b88f8b25e25e733c1a648'
print(bytes(signing_key.verify_key.to_curve25519_public_key()).hex()) # '1043ad34d3296c17cfe556e170452b6d896e52c4ed911f254711398e58476e20'
for n in range(0x41, 0x4a):
    signing_key = nacl.signing.SigningKey(bytes([n] * 32))
    print('{} -> {}'.format(bytes(signing_key).hex(), bytes(signing_key.verify_key).hex()))

4141414141414141414141414141414141414141414141414141414141414141 -> db995fe25169d141cab9bbba92baa01f9f2e1ece7df4cb2ac05190f37fcc1f9d
4242424242424242424242424242424242424242424242424242424242424242 -> 2152f8d19b791d24453242e15f2eab6cb7cffa7b6a5ed30097960e069881db12
4343434343434343434343434343434343434343434343434343434343434343 -> 22fc297792f0b6ffc0bfcfdb7edb0c0aa14e025a365ec0e342e86e3829cb74b6
4444444444444444444444444444444444444444444444444444444444444444 -> d759793bbc13a2819a827c76adb6fba8a49aee007f49f2d0992d99b825ad2c48
4545454545454545454545454545454545454545454545454545454545454545 -> 6355691c178a8ff91007a7478afb955ef7352c63e7b25703984cf78b26e21a56
4646464646464646464646464646464646464646464646464646464646464646 -> ee93a4f66f8d16b819bb9beb9ffccdfcdc1412e87fee6a324c2a99a1e0e67148
4747474747474747474747474747474747474747474747474747474747474747 -> e28a8970753332bd72fef413e6b0b2ef1b4aadda7aa2c141f233712a6876b351
4848484848484848484848484848484848484848484848484848484848484848 -> d4eec1869fb1b8a4e817516ad5a931557cb56805c3eb16e8f3a803d647df7869
4949494949494949494949494949494949494949494949494949494949494949 -> 772c8a442b7db06e166cfbc1ccbcbcde6f3eba76a4e98ef3ffc519502237d6ef
```

mcudev, 2021-01-16:
i was thinking about adding to `trezor-firmware/docs/hardware/model-t/boot.md` a description about how to grab a production bootloader and put it on a microsd card and use the boardloader to re-install it. something like:

First, you need to get a satoshilabs signed bootloader in order to run this process.

`wget https://wallet.trezor.io/data/firmware/2/trezor-2.3.4-bitcoinonly.bin`

after download, look at the firmware file in a hex editor and search for "TRZB". you'll see:
```
00087520: 5452 5a42 0004 0000 0000 0000 006c 0100  TRZB.........l..
00087530: 0200 0300 0200 0000 0000 0000 0000 0000  ................
00087540: cd0c 1493 8dab 0375 c754 7cf4 0874 57f9  .......u.T|..tW.
00087550: 9f77 62a1 dcf8 eb0a 774b 3f9b a267 8058  .wb.....wK?..g.X
00087560: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00087570: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00087580: 0000 0000 0000 0000 0000 0000 0000 0000  ................
```
look at https://github.com/trezor/trezor-firmware/blob/master/docs/hardware/model-t/boot.md#bootloader-format
from the hex dump above:
length of the bootloader header + length of the bootloader code (without the header) = 94208
0x87520 is decimal 554272, thus:

`dd if=trezor-2.3.4-bitcoinonly.bin bs=1 skip=554272 count=94208 of=trezor-bootloader.bin`
`sudo dd if=trezor-bootloader.bin of=/dev/mmcblk0 bs=512 conv=fsync`

unplug the Trezor T, insert the microsd card, and plug the Trezor T in.

doing this clears the writable flash and installs a bootloader that will put you into bootloader mode on the next restart.