trezor/trezor-firmware #695: Recovery: unplugging Trezor before clicking on Success leads to an incorrect state

tsusanka, 2019-11-11:
After I enter the last share of my Shamir Backup I get:

[image]

If I unplug the Trezor before clicking on "Continue" and start it again I get:

[image]

which is incorrect.

tsusanka, 2019-11-11:
We can either

1. Revert [this](https://github.com/trezor/trezor-firmware/commit/46c10e94021052989c66870c38290ec795bddd53). In another words end the recovery process before showing the success screen. Although this would probably now work with Wallet since we are now accepting all messages, it is a bit weird for the user - the Wallet gets refreshed before the user is able to click Continue.

2. Simly check somewhere in the homescreen if the remaining shares are 0 and identifier / exponent / backup type is set (probably just backup type is enough?) then we would end the progress.

@matejcik is what do you think?

tsusanka, 2019-11-11:
We have agreed on 1.

closed 2019-11-11 by tsusanka

bosomt, 2019-12-14:
now it says starting up and after that PIN NOT SET My Trezor.

So it works fine. Tested

commit 59727772 (2019-11-11): core/recovery: end progress before showing success

This fixes #695. This was the default behaviour but we have changed it
because of Wallet, which was doing the polling. Since Wallet can now
send RecoveryDevice again and acquire Context this is no more needed.