trezor/trezor-firmware #696: FIDO2 tests sometimes fail on CI

tsusanka, 2019-11-11:
FIDO2 test suite sometimes fail on CI. I'm not sure if this a tests problem or a CI problem. See this job: https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/347089653

We might add `retry` again but in a long run having each job doing a number of retries seems as a bad idea.

cc @1000101

1000101, 2019-11-11:
Indeed, this seems to be a CI issue (some scripts tend to fail when the node is running low on space). Should be fixed now.

tsusanka, 2019-11-11:
Rerun worked, so it seems to be fixed for now, thank you.

closed 2019-11-11 by tsusanka

tsusanka, 2020-04-16:
@1000101 could you please check the node is not running low on space again? This issue was different than what we are now experiencing but not _that_ different :).

1000101, 2020-04-16:
> @1000101 could you please check the node is not running low on space again? This issue was different than what we are now experiencing but not _that_ different :).

Yes, I can confirm the node is in a good shape. Moreover, a daily automatic pruning is in place (for quite some time now) so the probability of the node running out of space ever again is highly unlikely.