trezor/trezor-firmware #698: FIDO2: Ed25519 keys support

prusnak, 2019-11-11:
Let's implement Ed25519 keys in FIDO2.

prusnak, 2020-02-22:
The code is available in the [`andrewkozlik/fido2-ed25519`](https://github.com/trezor/trezor-firmware/tree/andrewkozlik/fido2-ed25519) branch.

andrewkozlik, 2020-02-22:
> The code is available in the [`andrewkozlik/fido2-ed25519`](https://github.com/trezor/trezor-firmware/tree/andrewkozlik/fido2-ed25519) branch.

Correct. But back in November I wasn't able to get it to work with webauthn.io, which was the only service supporting ed25519 in FIDO, that I could find. The problem is probably in the public key format.

prusnak, 2020-02-23:
We might check with the freshly released OpenSSH 8.2 which supports ed25519 FIDO2 keys.

prusnak, 2020-02-25:
I rebased the changes on current master => (`prusnak/fido2-ed25519`)(https://github.com/trezor/trezor-firmware/tree/prusnak/fido2-ed25519) branch.

I tried to enroll the key with the new OpenSSH 8.2 (`ssh-keygen -t ed25519-sk`) with the following result:

```
Key enrollment failed: invalid format
```

So it seems there is indeed something wrong with the public-key side of things for Ed25519 (`ssh-keygen -t ecdsa-sk` works fine with OpenSSH).

I think the `invalid format` error is being thrown from here: https://github.com/openssh/openssh-portable/blob/429170f273ce1b0140f8111a45ba69390d98de3a/ssh-ed25519-sk.c#L75 or here: https://github.com/openssh/openssh-portable/blob/429170f273ce1b0140f8111a45ba69390d98de3a/ssh-ed25519-sk.c#L102

andrewkozlik, 2020-02-26:
So I found a bug in my code, but after fixing it, I found a bug in libfido2: https://github.com/Yubico/libfido2/issues/135

andrewkozlik, 2020-02-26:
Looks like there is more: https://github.com/Yubico/libfido2/issues/136
@onvej-sl check to make sure I didn't overlook something.

andrewkozlik, 2020-02-27:
One more: https://github.com/Yubico/libfido2/issues/139

andrewkozlik, 2020-02-27:
I posted some concerns regarding COSE EdDSA public key format to their mailing list: https://mailarchive.ietf.org/arch/msg/cose/V9GXBGZ14xOdzml4GI2yF4McKkA/

andrewkozlik, 2020-02-27:
New branch [andrewkozlik/fido2-ed25519-2](https://github.com/trezor/trezor-firmware/tree/andrewkozlik/fido2-ed25519-2) works with https://webauthn.io.

prusnak, 2020-02-28:
I applied both of the changes in libfido2 and rebuilt openssh and I am still unable to enroll the ed25519-sk key.

```
$ ssh-keygen -t ed25519-sk
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
Key enrollment failed: invalid format
```

andrewkozlik, 2020-02-28:
Strange, it works like a charm for me. Are you sure you used the new firmware branch [andrewkozlik/fido2-ed25519-2](https://github.com/trezor/trezor-firmware/tree/andrewkozlik/fido2-ed25519-2)?
```
$ ./ssh-keygen -t ed25519-sk
Generating public/private ed25519-sk key pair.
You may need to touch your authenticator to authorize key generation.
Enter file in which to save the key (/home/andrew/.ssh/id_ed25519_sk): 
Enter passphrase (empty for no passphrase): 
Enter same passphrase again: 
Your identification has been saved in /home/andrew/.ssh/id_ed25519_sk
Your public key has been saved in /home/andrew/.ssh/id_ed25519_sk.pub
The key fingerprint is:
SHA256:zLhAzKXd5lrQgfsJ2hBOuHWGqlSvhzRzn7jyi39pkP8 andrew@dell
The key's randomart image is:
+[ED25519-SK 256]-+
|   . ....        |
|  .+=++o .       |
|  .**+o.+        |
| .o=oooB         |
|... B+=oS.       |
|.  o.*.*o        |
|    . * .        |
|   ... =         |
|   .+++ .E       |
+----[SHA256]-----+
```

prusnak, 2020-02-29:
@andrewkozlik I am on the following commits:

* https://github.com/trezor/trezor-firmware/commit/ddb06ed097576429fb201ce7d8643eeda93fdc57 (andrewkozlik/fido2-ed25519-2 branch)
* https://github.com/Yubico/libfido2/commit/0a44eff9c209765222056971b0793b35705b7a91 (master including your fixes)
* OpenSSH 8.2p1 from https://cdn.openbsd.org/pub/OpenBSD/OpenSSH/portable/openssh-8.2p1.tar.gz

closed 2020-03-12 by andrewkozlik

andrewkozlik, 2020-03-12:
@trezor/qa Please conduct general testing of FIDO2 registration/authentication with various combinations of sites, browsers and OS's. Try at least Google, Binance, GitHub and login.live.com (preferably others too) in each case. Check that credentials created with a previous firmware version can still be used for logging in.

andrewkozlik, 2020-03-12:
@trezor/qa I am planning on implementing https://github.com/trezor/trezor-firmware/issues/735 into before the upcoming freeze, if nothing more important comes up. So hold off on testing this until #735 is closed, so that you don't have to do the FIDO testing twice.

bosomt, 2020-03-24:
ssh-keygen -t ed25519-sk
works fine with TT
for T1 says: Key enrollment failed: invalid format

bosomt, 2020-03-24:
login.live.com is  probably broken

2.2.0 works fine
2.3.0 is able to create credential but login.live.com fails when saving token to database

im not able to check if it was saved to device, _trezorctl webauthn-list-credentials_ doesn't exist anymore

andrewkozlik, 2020-03-24:
> im not able to check if it was saved to device, _trezorctl webauthn-list-credentials_ doesn't exist anymore

There was a major overhaul of the trezorctl commands, about which I am not too happy myself. Here is the new form of the FIDO-related commands:

`trezorctl fido credentials list`
`trezorctl fido credentials add <HEX_CREDENTIAL_ID>`
`trezorctl fido credentials remove -i <index>`

bosomt, 2020-03-25:
MS Live is working fin in latest 8ae0535e buil

bosomt, 2020-03-25:
tested on login.gov, github, dropbox and so on and it works fine with 8ae0535e

sorooris, 2020-04-01:
QA OK - 2.3.0, revision 8ae0535e
tested login + reg on following OS' and browsers:
- Nixos - Chromium 79, Firefox 72
- Win10 - Chrome 80, Firefox 74, Edge 44

on following sites:
- WebAuthn.io
- Google
- GitHub
- login.live.com
- dropbox
- gandi.net

Firefox offered U2F only; Chrome and Chromium FIDO2 (excluding Google)

holycleugh, 2020-05-08:
@andrewkozlik, should I currently expect libfido2 to interpret "k->x" as the standard Ed25519 public key, then? I see it is getting passed directly to EVP_PKEY_new_raw_public_key.

andrewkozlik, 2020-05-08:
> @andrewkozlik, should I currently expect libfido2 to interpret "k->x" as the standard Ed25519 public key, then (rather than the x-coordinate as mentioned in RFC8152)? I see it is getting passed directly to EVP_PKEY_new_raw_public_key.

The current COSE specification is wrong. The "x" parameter of an octec key pair should be the standard Ed25519 public key as defined in [RFC8032](https://tools.ietf.org/html/rfc8032#section-5.1.2). The latest [draft](https://tools.ietf.org/html/draft-ietf-cose-rfc8152bis-algs-07#section-7.2) of the new COSE spec fixes the issue.