trezor/trezor-firmware #702: Decline button for a new/reset PIN

STew790, 2019-11-13:
Just a suggestion for setting a new PIN or resetting a current one.

When you try to decline the set/reset of the PIN, there should be a different button then a lock sign on a red background, for example, a cross.

The lock on a red background should be there only when you want to decline this when you are actually entering a PIN on a locked device.

firmware version 2.1.8

[image]

closed 2020-06-05 by andrewkozlik

tsusanka, 2020-06-16:
@trezor/qa please retest.

- When setting a new PIN the left lower button should not be a lock but an orange `<-`. Please check that and test that the functionality did not change.

bosomt, 2020-06-16:
2.3.1

[image]

vs

2.3.2 current master trezor-fw-regular-2.3.2-b6c8cbc2.bin

[image]

and yes functionality did not changed, you can set PIN code and you are asked for PIN code next time

tsusanka, 2020-06-16:
Great. Please add the QA test successful then please.

sorooris, 2020-06-16:
please ignore, i'm testing markdown

**QA OK**

2.3.1 8859d10 | 2.3.2 b6c8cbc2
------------ | -------------
[image] | [image]

Note: Functionality did not change - PIN code can be set is asked for the next time.

PR #1042 (2020-06-05): Fix cancel icon in PIN dialog
Use the Lock icon in the PIN entry dialog only when the device is actually locked.
Fixes #702.