trezor/trezor-firmware #703: Rename webauthn, u2f, etc. to "fido"

matejcik, 2019-11-13:
The functionality that was originally U2F and grew to include WebAuthN and FIDO2 is named ... inconsistently.

A good name for the whole complex is `fido`. This name should be used in protobuf messages, app/module names, etc.

This will be implemented in trezorlib/trezorctl. An open question is, should we rename it everywhere else too? Is it worth the hassle?

prusnak, 2019-11-13:
Places where a rename can/should take place:

- [ ] app name `core/src/apps/webauthn` -> `core/src/apps/fido`
- [ ] protobuf messages `{GetNext,Next,Set}U2FCounter` -> `{GetNext,Next,Set}FIDOCounter` (`messages-management.proto`)
- [ ] protobuf messages `WebAuthn*` -> `FIDO*` (`messages-webauthn.proto`)
- [ ] storage: `U2F_COUNTER` -> `FIDO_COUNTER`

matejcik, 2020-03-23:
> * [ ]  protobuf message names `{GetNext,Next,Set}U2FCounter` -> `{GetNext,Next,Set}FIDOCounter` (`messages-management.proto`)
> * [ ]  protobuf messages names `WebAuthn*` -> `FIDO*` (`messages-webauthn.proto`)

`*FIDOCounter` should be all moved to `messages-fido.proto` too