trezor/trezor-firmware #706: Investigate CSV usage

prusnak, 2019-11-13:
Some wallets (such as Blockstream Green) use CSV ([CHECKSEQUENCEVERIFY](https://github.com/bitcoin/bips/blob/master/bip-0112.mediawiki)) in their scripts. Let's ask them how their scripts look like and see how we can integrate this into our protobuf messages.

prusnak, 2019-11-13:
@romanz Can you help us with finding the right person to document how exactly does the scripts in Blockstream Green look like? (We don't like signing arbitrary scripts, so we have to understand the structure first and "teach" Trezor how to build transactions like these).

romanz, 2019-11-13:
Green wallet is using [libwally-core](https://github.com/ElementsProject/libwally-core) for generating the Bitcoin scripts for CSV-based multisig. There are currently two types of such scripts (where `main` key is controlled by the server and `recovery` keys are controlled by the user):
1. [(2-of-2) or (1-of-2 after CSV)](https://github.com/ElementsProject/libwally-core/blob/0041b04644687e0aac7c1611fe31c810babac66d/src/script.c#L663-L676):
```
OP_DEPTH OP_1SUB
OP_IF
  # The stack contains the main and and recovery signatures.
  # Check the main signature then fall through to check the recovery.
  <main_pubkey> OP_CHECKSIGVERIFY
OP_ELSE
  # The stack contains only the recovery signature.
  # Check the CSV time has expired then fall though as above.
  <csv_blocks> OP_CHECKSEQUENCEVERIFY OP_DROP
OP_ENDIF
# Check the recovery signature
<recovery_pubkey> OP_CHECKSIG
```
2. [(2-of-3) or (1-of-2 after CSV)](https://github.com/ElementsProject/libwally-core/blob/0041b04644687e0aac7c1611fe31c810babac66d/src/script.c#L718-L734)
```
OP_DEPTH OP_1SUB OP_1SUB
OP_IF
  # The stack contains 3 items, a dummy push for the off-by-one bug
  # in OP_CHECKMULTISIG, and any 2 of the 3 signatures.
  OP_2 <main_pubkey>
OP_ELSE
  # The stack contains a dummy push as above, and either of the
  # recovery signatures.
  <csv_blocks> OP_CHECKSEQUENCEVERIFY OP_DROP
  # Note OP_0 is a dummy pubkey that can't match any signature. This
  # allows us to share the final OP_3 OP_CHECKMULTISIGVERIFY case
  # thus reducing the size of the script.
  OP_1 OP_0
OP_ENDIF
# Shared code to check the signatures provided
<recovery_pubkey_1> <recovery_pubkey_2> OP_3 OP_CHECKMULTISIG
```

There is a proof-of-concept [trezor-firmware branch](https://github.com/trezor/trezor-firmware/compare/master...romanz:csv-multisig) that allows the user to create p2sh & bech32 addresses controlled by the first type of [CSV-based multisig script above](https://github.com/romanz/trezor-firmware/blob/dead10c01e4871c894ffbb51539ffb885f38c511/core/src/apps/wallet/sign_tx/scripts.py#L271-L300) and spend from them.
I've added a `csv` field to the [multisig protobuf definition](https://github.com/romanz/trezor-firmware/blob/dead10c01e4871c894ffbb51539ffb885f38c511/common/protob/messages-bitcoin.proto#L31) which causes the code to use CSV-based multisig script (instead of the OP_CHECKMULTISIG "standard" one).
Although It's a bit "hacky", it seems to work as intended (see integration tests for Bitcoin [p2sh](https://github.com/romanz/trezor-firmware/blob/dead10c01e4871c894ffbb51539ffb885f38c511/tests/device_tests/test_msg_signtx_segwit.py#L426-L747) & [bech32](https://github.com/romanz/trezor-firmware/blob/dead10c01e4871c894ffbb51539ffb885f38c511/tests/device_tests/test_msg_signtx_segwit_native.py#L814-L1016) addresses and the [following Testnet transaction](https://blockstream.info/testnet/tx/0cc9159979949c2324907e6a9fb8afeb79be51e6fa5b65e4014eb03b38a1f3ba?expand) for cooperative spending example from CSV-based multisig address).

prusnak, 2019-11-15:
Thanks for the code! We'll have a look at it

prusnak, 2019-12-16:
I agree it makes sense to do this, but we need to refactor the `wallet.sign_tx` app first: https://github.com/trezor/trezor-firmware/issues/617

romanz, 2019-12-16:
Sounds good, thanks for the update!

romanz, 2020-01-18:
Updated and rebased [`csv-multisig`](https://github.com/trezor/trezor-firmware/compare/master...romanz:csv-multisig) branch over the latest `master`.

romanz, 2020-02-13:
Rebased over latest master and force-pushed https://github.com/trezor/trezor-firmware/commit/95979fc9ff7fe312e80280d5f75a199153e640b7.

tsusanka, 2020-08-03:
Related to #416.

landabaso, 2022-02-23:
Sorry for the pseudo off-topic.
What is the reason @greenaddress (and other CSV wallets) use <pub keys> instead of <pub key hashesz>?
Wouldn't the output scripts be smaller in size if "DUP HASH160 <pubKHash> " was used?

Georgantas, 2022-05-19:
Also the miniscript policy for a 2-of-2 that turns into a 1-of-2 after 90 days: `and_v(v:pk(key_user),or_d(pk(key_service),older(12960)))` (from: https://bitcoin.sipa.be/miniscript/ )