trezor/trezor-firmware #707: vendor/fido2 submodule commit not found on master

tsusanka, 2019-11-14:
I'm really confused about this. Running: `git submodule update --init --recursive --force` in the repository root leads to:

```
error: Server does not allow request for unadvertised object e1e44d43893fd8eefde4f18f576d006747e5fc90
Fetched in submodule path 'vendor/fido2-tests', but it did not contain e1e44d43893fd8eefde4f18f576d006747e5fc90. Direct fetching of that commit failed.
```

If I `cd` into `vendor/fido2-tests` I indeed can't find the commit:

```
$ cd vendor/fido2-tests
$ git fetch
$ git checkout e1e44d43893fd8eefde4f18f576d006747e5fc90
fatal: reference is not a tree: e1e44d43893fd8eefde4f18f576d006747e5fc90
```

but on github I can see it: https://github.com/trezor/fido2-tests/commit/e1e44d43893fd8eefde4f18f576d006747e5fc90 and based on .gitmodules I am indeed using https://github.com/trezor/fido2-tests.git

matejcik, 2019-11-14:
try using `git submodule deinit --all` first?

tsusanka, 2019-11-14:
No success:

```
$ git submodule deinit --all -f
Cleared directory 'common/defs/ethereum/tokens'
warning: Could not unset core.worktree setting in submodule 'common/defs/ethereum/tokens'
Cleared directory 'crypto/tests/wycheproof'
warning: Could not unset core.worktree setting in submodule 'crypto/tests/wycheproof'
Cleared directory 'vendor/QR-Code-generator'
warning: Could not unset core.worktree setting in submodule 'vendor/QR-Code-generator'
Cleared directory 'vendor/fido2-tests'
Submodule 'vendor/fido2-tests' (https://github.com/trezor/fido2-tests.git) unregistered for path 'vendor/fido2-tests'
Cleared directory 'vendor/libopencm3'
Submodule 'legacy/libopencm3' (https://github.com/libopencm3/libopencm3.git) unregistered for path 'vendor/libopencm3'
Cleared directory 'vendor/micropython'
Submodule 'vendor/micropython' (https://github.com/trezor/micropython.git) unregistered for path 'vendor/micropython'
Cleared directory 'vendor/nanopb'
Submodule 'legacy/vendor/nanopb' (https://github.com/nanopb/nanopb.git) unregistered for path 'vendor/nanopb'
Cleared directory 'vendor/secp256k1-zkp'
Submodule 'vendor/secp256k1-zkp' (https://github.com/ElementsProject/secp256k1-zkp.git) unregistered for path 'vendor/secp256k1-zkp'
$ git submodule update --init --recursive --force
Submodule 'common/defs/ethereum/tokens' (https://github.com/ethereum-lists/tokens.git) registered for path 'common/defs/ethereum/tokens'
Submodule 'crypto/tests/wycheproof' (https://github.com/google/wycheproof) registered for path 'crypto/tests/wycheproof'
Submodule 'vendor/QR-Code-generator' (https://github.com/nayuki/QR-Code-generator.git) registered for path 'vendor/QR-Code-generator'
Submodule 'vendor/fido2-tests' (https://github.com/trezor/fido2-tests.git) registered for path 'vendor/fido2-tests'
Submodule 'legacy/libopencm3' (https://github.com/libopencm3/libopencm3.git) registered for path 'vendor/libopencm3'
Submodule 'vendor/micropython' (https://github.com/trezor/micropython.git) registered for path 'vendor/micropython'
Submodule 'legacy/vendor/nanopb' (https://github.com/nanopb/nanopb.git) registered for path 'vendor/nanopb'
Submodule 'vendor/secp256k1-zkp' (https://github.com/ElementsProject/secp256k1-zkp.git) registered for path 'vendor/secp256k1-zkp'
Submodule path 'common/defs/ethereum/tokens': checked out '4dc60b7964b655a2c2186782c8b40a06f68221a5'
Submodule path 'crypto/tests/wycheproof': checked out '2904be69e9d666bf3064fdc15093747e695cfae6'
Submodule path 'vendor/QR-Code-generator': checked out '40d24f38aa0a8180b271b6c88be8633f842ed9d4'
error: Server does not allow request for unadvertised object e1e44d43893fd8eefde4f18f576d006747e5fc90
Fetched in submodule path 'vendor/fido2-tests', but it did not contain e1e44d43893fd8eefde4f18f576d006747e5fc90. Direct fetching of that commit failed.
```

prusnak, 2019-11-14:
Resolved by:
```
git submodule sync
cd vendor/fido2-tests ; git fetch
cd ../..
git submodule update --init --recursive
```

closed 2019-11-14 by prusnak