trezor/trezor-firmware #708: Export script improvement

ZdenekSL, 2019-11-14:
Child of #418 

This issues have 2 parts
1. Update the script so it shows the assignee of the issue
2. Update the script so it can export issues from private repositories, mainly from https://github.com/satoshilabs/buytrezor

tsusanka, 2019-11-14:
I have added assignee in https://github.com/trezor/trezor-firmware/commit/e38d6ca401f679985f0a865c3943be763fa907a1 . But bear in mind, that the GitHub API offers bot `assignee` and `assignees` and I'm not sure how it selects the assignee. But in case we would print assignees in one field you couldn't sort it properly anyway, so this is what you want probably?

tsusanka, 2019-11-14:
Private repositories support added in https://github.com/trezor/trezor-firmware/commit/56e84aa0e67aeb237456beebfa22e71ceeabb213. Don't forget that issues without weights are being skipped.

You need to add a personal access token to a file you create yourself called `github_issues_to_csv.ignore` and placed it in the same directory as the script is. In this file you need to insert a personal access token you create on github, see this [tutorial](https://help.github.com/en/github/authenticating-to-github/creating-a-personal-access-token-for-the-command-line).

closed 2019-11-14 by tsusanka