trezor/trezor-firmware #715: RecoveryDevice never announces invalid seed to host

matejcik, 2019-11-19:
An interesting side-effect of the new recovery homescreen is that the host never finds out if the user enters an invalid seed. The user is notified on screen, but only has the option to try again or cancel the process.

Previously (and still on T1):

Host->Trezor: RecoveryDevice
User: "stick stick stick..." * 12
Trezor->Host: Failure("Invalid seed")

Now (with recovery homescreen):

Host->Trezor: RecoveryDevice
User: "stick stick stick..." * 12
Trezor (on screen): Invalid seed. Try again.
User: "all all all..." * 12
Trezor->Host: Failure("Seed is valid but does not match device.")

IMO the new behavior is OK, but let's look at it a little and see if there are any implications.

I am not sure if there is a way to "fix" this. We could abort the recovery flow completely if the user enters a wrong BIP39 seed, but that doesn't seem better in any way. Alternately, the "try again" screen could have a "cancel" button right on it, and pressing it would return the appropriate failure to the Host, but that's sort of weird UX-wise.

tsusanka, 2019-11-21:
> Trezor->Host: Failure("Seed is valid but does not match device.")

We return this incorrect message in case of recovery as well or that's a mistake?

> Alternately, the "try again" screen could have a "cancel" button right on it, and pressing it would return the appropriate failure to the Host, but that's sort of weird UX-wise.

That sounds okay to me.

matejcik, 2019-11-21:
> > Trezor->Host: Failure("Seed is valid but does not match device.")
> 
> We return this incorrect message in case of recovery as well or that's a mistake?

Not sure what you're asking. The message is correct in this case, because the user entered a valid seed on the second try (but one that I, as the example author, chose to be different from the loaded one ;) )

> 
> > Alternately, the "try again" screen could have a "cancel" button right on it, and pressing it would return the appropriate failure to the Host, but that's sort of weird UX-wise.
> 
> That sounds okay to me.

I wouldn't want to implement this, unless we have a compelling use case to send the information to host. The flow where the user retries or cancels (and therefore the response is "cancelled") seems more correct.

matejcik, 2020-06-22:
Beta of Suite is out and nobody complained about this so far. It doesn't seem that this feature is necessary.

closed 2020-06-22 by matejcik