trezor/trezor-firmware #727: Add wipe code activation device test for T1

andrewkozlik, 2019-11-25:
I wrote a device test for T2 which checks that when the wipe code is entered instead of the PIN, it will cause the device to wipe itself. The T2 emulator displays a popup for 5 seconds, calls `shutdown()` and automatically restarts. So I simply let the device test wait for 7 seconds before continuing, see https://github.com/trezor/trezor-firmware/pull/720/commits/89a3ec16965d80e4cc3b946a6d364f3a05ba51fc#diff-9f7414700342058eee129030ac8f0d30R253. I am assuming that waiting 2 seconds longer than the length of the popup is enough. Perhaps @matejcik knows of a better solution. Maybe polling the emulator to see if it's ready?

Adding the same test for T1 is more complicated, because the T1 emulator doesn't restart after `shutdown()`. I attempted two solutions:
1. Automatically restart emulator using `longjmp()`, see https://github.com/trezor/trezor-firmware/commit/7b0bb93767c9f90fe3a805f1c0d0f35e93d4038f. This appears to be working fine, but it might easily break, because the legacy code has lots of static variables, which are hard to keep track of and reset, such as `read_state` https://github.com/trezor/trezor-firmware/commit/7b0bb93767c9f90fe3a805f1c0d0f35e93d4038f#diff-aa94d6284fda38c026da1e6e45394cb4R229.
2. Simply keep restarting the emulator whenever it shuts down in `legacy/script/test`, see https://github.com/trezor/trezor-firmware/commit/6dcb3f423033fc3cd62114b1eb82123a9a60d290. This isn't working yet, probably because the test suite needs to reconnect to the emulator after restart.

Any ideas @prusnak?

prusnak, 2019-11-26:
What exactly causes that T2 emulator is restarted?

andrewkozlik, 2019-11-26:
> What exactly causes that T2 emulator is restarted?

In the emulator the C function `__shutdown()` calls `sys.exit(3)` which raises `SystemExit` (this is derived directly from `BaseException`). I hadn't investigated this further and simply assumed that this exception gets intentionally caught somewhere to restart the emulator, but now that I look at it more closely, I am not sure that this behavior is intentional. The place where the exception gets caught is https://github.com/trezor/trezor-firmware/blob/master/core/src/trezor/wire/__init__.py#L408, so it's not really restarting the emulator.

matejcik, 2020-02-04:
good catch, in particular i'm not too convinced that the `error_shutdown` function should convert to `sys.exit` ([these lines](https://github.com/trezor/trezor-firmware/blob/75264a07a883dc1c6494442b9e3bf28f08fc021e/core/embed/unix/main.c#L417-L422) shouldn't be there IMO)?

the `fflush()` is a nicety of course, but other than that, what sort of cleanup do we expect micropython to do?

matejcik, 2020-02-04:
WRT the original problem: the wipe-code test should shut down the emulator, and so it should be in the part of the test suite where we do controlled emulator tests. We should also assert that the emulator actually shuts down.

PR #840 (2020-02-13): wipe code activation tests
this adds wipe code activation test for T1 and TT (fixes #727)

in relation, the emulator restart issue is resolved (fixes #826)

i have also simplified the wipe code setup test for T1 with a new debuglink feature
closed 2020-02-13 by matejcik