trezor/trezor-firmware #731: All zeroes when using multiple Trezors as U2F in Firefox

matejcik, 2019-11-28:
to reproduce:

1. plug in two TTs
2. open Firefox web browser and go to https://u2f.bin.coffee/
3. click "U2F Register"
4. both Trezors will show a pop-up with the website name
5. click Accept on one of them
6. click "U2F Sign"

result:
Trezor A (the one that was registered) shows "authenticate"
Trezor B (the one that was NOT registered) shows "register" for app name "0000....0000"

expected:
in Chrome, Trezor B will show a "not registered - another device was registered" pop-up

---

another issue with similar symptoms:

1. plug in one TT and register it as a U2F device on e.g. Facebook or Google
2. plug in second TT and go to Key management
3. click "Add new security key"

result:
Trezor A that was previously registered will show "register" for app name "0000....0000"
Trezor B that was NOT registered will show the proper "register" dialog, with name and icon

expected:
Trezor A that was previously registered shouldn't do anything? (this is what happens in Chrome)

matejcik, 2019-11-28:
[image]

andrewkozlik, 2019-12-03:
When browsers want to make U2F authenticators blink and wait for user presence they send a fake registration request. Firefox uses a bogus application ID `\x00\x00 ... \x00` for this purpose and Chrome uses `\x41\x41 ... \x41`. These values do not come from any official specs. Sometimes the reason is that the device is _not registered_ with the RP and other times the reason is that the device is _already registered_ with the RP. There is absolutely no way to tell the difference from the given request itself. However, it is possible to deduce the reason from previous U2F_AUTHENTICATE check-only requests on the same channel. One of the commits in https://github.com/trezor/trezor-firmware/compare/andrewkozlik/731?expand=1 implements this solution.

closed 2019-12-04 by andrewkozlik

andrewkozlik, 2019-12-04:
@trezor/qa 
I have tested this only in Firefox and Chrome on Ubuntu. The device should now show a "Not registered" error or "Already registered" error accordingly, instead of 0000...0000. Please retest the two scenarios described above and in addition test with other browsers and OSs (Chrome/Firefox/Edge/Safari on Windows and on Mac).

bosomt, 2019-12-04:
@andrewkozlik FW ? 2.1.8 ? i wasn't able to reproduce it last weekend with mix of 2.1.8 and 2.1.9

bosomt, 2019-12-04:
i was able to reproduce it with 2xTT woth 2.1.8 FW
with latest 2.1.9 it says not registered

tsusanka, 2019-12-04:
@bosomt what is latest 2.1.9 for you? Did you download a new one from `master`? Because @andrewkozlik did some fixes today and yesterday.

bosomt, 2019-12-04:
from here: https://gitlab.com/satoshilabs/trezor/trezor-firmware/pipelines

bosomt, 2019-12-04:
but there is new issue ;) 
i registered both and now when i press sign one of them randomly says not registered.

tsusanka, 2019-12-04:
> from here: https://gitlab.com/satoshilabs/trezor/trezor-firmware/pipelines

Great. If you got one from master after @andrewkozlik closed this issue it should be the correct one (should be 20cda137b7399e2b2496b1824c687ea4744ba1fa probably).

> i registered both and now when i press sign one of them randomly says not registered.

If you have found a new issue, wait for it, create a new issue!

bosomt, 2019-12-04:
no new issue, false alarm ;)