trezor/trezor-firmware #733: Display light blinking when U2F via Dropbox is triggered on Andorid

bosomt, 2019-12-01:
Please see video
Same behavior with 2.1.8 and 2.1.9 firmware.
When you enable labeling and Dropbox asks for access to USB device Trezor light will go crazy till you reconnect it.

https://youtu.be/ZXDc-IIxesA

prusnak, 2019-12-01:
My hunch is that Android is retarded and keeps pushing in invalid U2F messages or something similar.

I am leaning to deferring this and resolving only if we the same issue with Suite on Android.

bosomt, 2019-12-02:
there is no labeling in suite yet so I'm not able to check if it works fine in suite.
If you need anything to follow this issue please let me know

bosomt, 2019-12-02:
works fine with 2.1.4 FW

andrewkozlik, 2019-12-02:
I implemented support for the CTAPHID_WINK command as part of the FIDO2 effort. The way it works is it calls ui.alert() which quickly flashes the screen three times. It looks like Android is permanently sending a huge number of CTAPHID_WINK commands to the device and they are overlapping each other which makes it blink so chaotically.

I implemented a possible fix (untested) which does the following:
1. Remember the CID of the last WINK command and ensure that we do only one WINK at a time on any given CID. 
2. If an alert is already in progress, then any further alert requests will be ignored to avoid multiple alerts overlapping.

closed 2019-12-04 by andrewkozlik

pankrats, 2019-12-18:
tested with fw 2.2.0 ver 6f0ee0d2 looks good, no problem with enabling Dropbox in wallet.

PR #734 (2019-12-04): Fix continuous display blinking with Android
Fixes #733. I haven't tested this myself, but @bosomt has.