trezor/trezor-firmware #735: FIDO2: Add maxCredentialCountInList to authenticatorGetInfo response

andrewkozlik, 2019-12-02:
Based on Chromium source code https://cs.chromium.org/chromium/src/device/fido/device_response_converter.cc?l=367&rcl=3344e8a5450bed6243b95148f39a0956763dd511, it looks like the CTAP spec will soon add two new members to the authenticatorGetInfo response, maxCredentialIdLength (0x07) and maxCredentialCountInList (0x08). Let's keep an eye out for the spec updates and add this when it appears in the spec. This might stop the clients from sending silent authentication requests.

prusnak, 2019-12-04:
It seems the code says it differently (i.e. you swapped the values):
*  `0x07` - `maxCredentialCountInList`
*  `0x08` - `maxCredentialIdLength`

andrewkozlik, 2020-03-20:
Tests cases: same as https://github.com/trezor/trezor-firmware/issues/698#issuecomment-598234338

closed 2020-03-20 by andrewkozlik

sorooris, 2020-04-01:
QA OK
see https://github.com/trezor/trezor-firmware/issues/698#issuecomment-607194611