trezor/trezor-firmware #737: FR: python-u2flib-host Trezor support

EDmitry, 2019-12-03:
python-u2flib-host (https://github.com/Yubico/python-u2flib-host) is a library initially written by Yubico to interface with U2F devices. It's used by other software products, more notably Qubes OS. Right now there is no support for Trezor. I have tried adding a USB vendor_id/product_id in the list of supported devices, but even though Trezor is recognized correctly, it keeps asking for the U2F confirmation that doesn't seem to go through. I suspect there is a slight difference in how Trezor handles U2F protocol versus other U2F compatible devices, that's why I'm creating an issue here and not in python-u2flib-host project for someone to verify this.

prusnak, 2019-12-04:
I was able to reproduce the issue with the following code:

``` python
import json
from u2flib_host import u2f, exc
from u2flib_host import hid_transport

hid_transport.DEVICES = [
    (0x1209, 0x53c1),
]

devices = u2f.list_devices()

print("devices", devices)

VERSION = 'U2F_V2'
FACET = 'https://example.com'
CHALLENGE = 'challenge'
KEY_HANDLE = b'\00' * 64

REG_DATA = json.dumps({
    'version': VERSION,
    'challenge': CHALLENGE,
    'appId': FACET
})

for device in devices:
    with device as dev:
        registrationResponse = u2f.register(dev, REG_DATA, FACET)
```

running this will result in the following output (for both T1 and TT) with stalled U2F dialog on the device screen:

```
devices [<u2flib_host.hid_transport.HIDDevice object at 0x7f0c18ed8c90>]
Traceback (most recent call last):
  File "foo.py", line 26, in <module>
    registrationResponse = u2f.register(dev, REG_DATA, FACET)
  File "/usr/lib/python3.7/site-packages/u2flib_host/u2f.py", line 69, in register
    return lib.register(device, data, facet)
  File "/usr/lib/python3.7/site-packages/u2flib_host/u2f_v2.py", line 73, in register
    response = device.send_apdu(INS_ENROLL, p1, p2, request)
  File "/usr/lib/python3.7/site-packages/u2flib_host/device.py", line 112, in send_apdu
    raise exc.APDUError(status)
u2flib_host.exc.APDUError: 0x6985
```

`u2flib_host.__version__` is `3.0.3`

prusnak, 2019-12-04:
Seems this is intentional and correct: https://github.com/Yubico/python-u2flib-host/issues/15

Please test your setup with Yubikey or at least provide the code you are using, so we can use the same code to debug.

EDmitry, 2019-12-04:
My specific use case is in Qubes OS using U2F proxy. I think it might not be polling the device. It seems a bit odd to me that the burden of differentiating between Yubikeys and devices that require the confirmation is on the client of the library. I understand that the protocol is synchronous, but I'm not sure why the library has to be as well.

prusnak, 2019-12-04:
OK, i am closing this as it seems there is nothing wrong on our side (i.e. you can't confirm this works with Yubikey). If there is any new info that says otherwise (and we should fix something on our end), please reopen the issue.

Btw, I suggest the change to fido2 libraries in the Qubes OS issue you linked to.

closed 2019-12-04 by prusnak

prusnak, 2019-12-04:
Raised issue for adding Trezor into `python-u2flib-host`: https://github.com/Yubico/python-u2flib-host/issues/42

EDmitry, 2019-12-04:
Thank you, Pavol!