trezor/trezor-firmware #741: Parse timestamp to datetime

tsusanka, 2019-12-05:
In Stellar and most likely in some other apps as well the host is sending a timestamp to Trezor. We currently do not have a function to format it into something more readable. Since we probably (?) do not want to include the whole datetime micropython library we can either:

1. use C and its time.h library (implemented in `tsusanka/timestamp`, but it fails on `multiple definition of snprintf`)
2. write it manually
3. add the datetime library

prusnak, 2019-12-05:
[datetime.py](https://github.com/micropython/micropython-lib/blob/master/datetime/datetime.py) in case we want to go with option 3

prusnak, 2019-12-09:
Another place where we could use this function is here:

https://github.com/trezor/trezor-firmware/blob/3d50bb5ed8ddb6025e4492731fbeaaa4649e23d9/core/src/apps/wallet/sign_tx/layout.py#L92 (when `lock_time` is interpreted as timestamp and not the blockheight).

prusnak, 2019-12-09:
Code in C: https://stackoverflow.com/questions/20826847/convert-between-date-time-and-time-stamp-without-using-std-library-routines

bosomt, 2021-11-17:
**QA OK**

fw trezor-fw-regular-2.4.3-221977ad.bin
fw trezor-fw-regular-1.10.4-221977ad.bin