trezor/trezor-firmware #747: Add use_self_attestation flag to FIDO apps

andrewkozlik, 2019-12-11:
Based on user feedback and experimenting I found that Microsoft Azure AD doesn't allow registering security keys which use self attestation, so we need to use basic attestation for login.microsoft.com. The reason why we had chosen to use basic attestation is that previous experiments had shown that Binance won't allow us to use basic attestation. The solution is to add a use_self_attestation flag which will allow us to control this on a per-app basis similar to the use_sign_count flag.

andrewkozlik, 2019-12-11:
PR: https://github.com/trezor/trezor-firmware/pull/746

prusnak, 2019-12-14:
Closed via https://github.com/trezor/trezor-firmware/pull/746

closed 2019-12-14 by prusnak

andrewkozlik, 2019-12-16:
@trezor/qa 
Check that FIDO2 registration and authentication work correctly for `login.live.com`, `binance.com` and `github.com`.
If you feel up to it, then you can also test FIDO2 with Azure AD. See https://ronnydejong.com/2019/07/12/microsoft-keeps-its-password-less-promise-and-ships-native-fido2-support-to-windows-10-azure-ad/