trezor/trezor-firmware #749: Idea: Visualize which passphrase is currently being used

tsusanka, 2019-12-12:
Upon discussion about passphrases with @szymonlesisz we came up with an idea to somehow let the user know which passphrase is being currently used. We obviously do not want to display the passphrase on the device in plain text.

We could use so-called identicons:

[image]

These identicons would be derived deterministicly, probably from the hash of a Bitcoin Testnet address. Then both Suite and Trezor could display this image and the user would be aware which passphrase they are using (or in another words, have a confirmation they use the one they expect). On Trezor we would either squeeze it on the homescreen or do a popup with the image on Intialize. To be decided, it is an UI challenge.

-----

Examples of identicon libraries:
- https://github.com/dmester/jdenticon
- https://github.com/download13/blockies (similar ones are used in Ethereum for address visualisation)
- https://github.com/stewartlord/identicon.js (similar ones are used in GitHub for users without avatars)

prusnak, 2020-02-16:
We agreed the solution needs to come from the Product team.

tsusanka, 2020-09-03:
Since the latest release we display the passphrase that's being used. Although you can still use more passphrases at the same time by using the sessions I don't think it makes sense to complicate the UI more. Closing?

prusnak, 2020-09-03:
Yes, closing ... I just wanted for the september update to be out there publicly.

closed 2020-09-03 by prusnak