trezor/trezor-firmware #757: Internal error when Wipe PIN code entered

bosomt, 2019-12-17:
Internal error when Wipe PIN code entered.

FW version 2.2.0

Chromium
Nixos
Beta wallet + trezorctl latest github build

How to preproduce

set wipe code
set PIN code
change PIN code
enter wipe PIN code 

or

set wipe code
set PIN code
start dry run
enter wipe PIN code 

expected: Device is wiped and you will see info that it was wiped

[image]

bosomt, 2019-12-17:
same behavior when you try to disable wipe code
trezorctl set wipe-code -r
you are asked for pin code, enter wipe pin code

internal error

prusnak, 2019-12-19:
Resolved in master + release/2020-10 branch.

closed 2019-12-19 by prusnak

bosomt, 2019-12-19:
works fine now