trezor/trezor-firmware #767: BUG: Super Shamir does not display warning when entering share multiple times

wendys-cats, 2019-12-19:
**Describe the bug**
When in recovery mode/dry run of Super Shamir Secret Sharing, the system does not display a warning in case you enter the same share multiple times. This ONLY applies to shares in the first group.
*Note:* This is a display issue only, as the system cannot use the same share more than once for solving the SuperSSS seed and it will not count it more than once when recovering the group, so entering it 7 times does bring the user closer to solving than entering it only once.

**Firmware version**
FW 2.20. and 2.1.8., possibly older

**Desktop/smartphone setup (please complete the following information):**
 - Trezor Wallet
 - OS: Linux NixOS
 - Browser Chromium 
 - Version 78.0.3904.108 (Developer Build) (64-bit)
 - Trezor model T device

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new wallet using Super Shamir Secret Sharing  trezorctl reset-device -b advanced)
2. Go to mytrezor.corp.sldev.cz/develop/development, tab Advanced (right now, dry run recovery does not work on 2.2.0. with the production wallet, so either do it here or use 2.1.8. FW)
3. Do a dry run, enter the share(s) from first group multiple times
4. Wipe device, restore the wallet and apply step 2 again

**Expected behavior**
After entering the fourth word of any share, system determines whether or not you have already entered this exact share before. If so, it will display a warning on the Trezor T device and asks that you enter another, unused share. It will also abort the operation, returning to displaying the current status of your recovery process.

tsusanka, 2019-12-19:
Could you provide me the shares you have used if public?

---

Typo in your "note": 

> ...it will not count it more than once when recovering the group, so entering it 7 times does **not** bring the user closer to solving than entering it only once.

Right? Just making sure.

wendys-cats, 2019-12-19:
@tsusanka here you go!
groups: 3/5
set size: 
1:4/6
trash smug adjust aide benefit temple round clogs devote prevent type cards clogs plastic aspect paper behaviour lunar custody intimate 
trash smug adjust ambition criminal prisoner security math cover pecan response pharmacy center criminal salary elbow bracelet lunar briefing dragon 
trash smug adjust arena beard quick language program true hush amount round geology should training practice language diet order ruin
trash smug adjust beam brave sack magazine radar toxic emission domestic cradle vocal petition mule toxic acid hobo welcome downtown
trash smug adjust black ajar squeeze often ajar company harvest view emperor username leaf glasses trash dwarf reward parcel frozen
2:3/7
trash smug being acne bracelet simple improve dominant aviation priest finger radar explain aspect teaspoon spew percent system kidney scramble
trash smug being agree darkness ancestor angel shaft morning formal flash alcohol equation jacket smart spirit romp thumb terminal purple
trash smug being amazing distance expect family pile kitchen union early pregnant knife blue relate repair unfold plot device gross
trash smug being arcade activity meaning credit fawm tracks broken drink capital junk gather museum rescue making merchant predator become
trash smug being axle database premium prevent syndrome warn scramble exact emission swing satoshi document picture bundle identify eyebrow
trash smug being bishop boundary holiday spine curly glimpse demand exotic vocal suitable random blimp making saver charity unknown company
trash smug being breathe acrobat course math elegant mama loan clothes frequent unusual sunlight gesture timber undergo humidity dish verify 
3:3/5
trash smug chemical agree angel party frozen award marathon already mouse lily cargo lunch likely taste resident valid favourite lily
trash smug chemical amazing early voter hobo society shadow dielmma diagnose climate scandal spray flame lunar camera steady likely security
trash smug chemical arcade advance champion gross rapids diminish pile racism friendly mule bracelet decorate pupal retreat usual exotic brave
trash smug chemical axle demand relate lend standard exact muscle retreat theater coding humidity romantic magazine camera capture photo owner 
4:5/8
trash smug demand acrobat bulb physics best parcel remind tendency morning pickup umbrella jump phantom fortune sack forbid picture visitor
trash smug demand aircraft ambition muscle beyond mother moring voting belong shelter vexed ajar inmate jewelry knit database element careful
trash smug demand amount dictate repeat curly genre else uncover radar aluminium angel shelter ting evening profile scramble slow main
trash smug demand argue detect taste alarm adequate infant screw axle mayor dictate dwarf space spelling advocate practicea seafood scramble 
trash smug demand beard depend tricycle peasant burning formal being pleasure spend aspect liquid pile rapids trust cdowntown pencil together
trash smug demand blanket acid early extend plan deploy making likely criminal elevator ting viral domain debut ambition plot enforce
trash smug demand broken briefing guest tactics wrist solution undergo famous anxiety step burden satoshi mineral erode steady warn escape
trash smug demand camera domain training album building climate entrance focus capacity provide vampire inmate picture legs verdict building exclude
5:2/3
trash smug elder acid angel paid decrease crazy wits quick force cubic capacity heat emerald swing spew ivory fangs username
trash smug elder agency already expect shadow unusual making solution liquid dress multiple mouse negative filter science triumph daughter tension

tsusanka, 2019-12-19:
Note to myslef: The words were manually copied so behaviour -> behavior, favourite -> favorite, dielma -> dilemma.

tsusanka, 2019-12-19:
Confirming, it indeed seems that after loading the device with shares above, then starting dry run and entering:

- "trash smug adjust aide benefit temple round clogs devote prevent type cards clogs plastic aspect paper behavior lunar custody intimate"
- "trash smug adjust ambition criminal prisoner security math cover pecan response pharmacy center criminal salary elbow bracelet lunar briefing dragon"

And then entering the second one again I do not get any warning whatsoever.

tsusanka, 2019-12-19:
Fixed in 6594ee93 (what a classic error!). I will add a test for this soon:tm: in #770, we should test this.

tsusanka, 2019-12-19:
@wendys-cats could you retest with the newest fw? https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/384964280/artifacts/download

----

@prusnak it is a tiny error, we do not need to squeeze it into the release imho. But it is basically a one-liner. What do you think?

closed 2019-12-19 by tsusanka

prusnak, 2019-12-19:
Let's backport into the release branch please

bosomt, 2019-12-21:
tested
fixed