trezor/trezor-firmware #769: single command to run all automated checks

matejcik, 2019-12-19:
Our CI pipeline is quite rich, but not available to external contributors, and it is difficult to run all tests at once.

It would be helpful to have a single command / make target that can:
* run all style checks,
* build both core and legacy emulators,
* build both core and legacy firmwares (optionally with/without BITCOIN_ONLY)
* run unit tests, device tests, click tests, upgrade tests, both for core and legacy

Having this execute in our Docker environment seems to be appropriate, so that we don't need to get people to install all dependencies.

vdovhanych, 2021-11-29:
@matejcik What about doing this with github actions, just extract some of our gitlab CI and allow it to run on gh actions on external prs. We already have some of the check here so it would mean only adding the emu and fw builds with some tests.

matejcik, 2021-11-29:
The original issue is about running the checks _locally_, i.e., something like `make ci` that you can run and it does everything.

Extending github actions could presumably fulfill the same purpose, but it feels like a different issue -- plus, we'd need to copy over basically the whole Gitlab pipeline?