trezor/trezor-firmware #778: Unable to use the --expand option in recovery mode

sako0938, 2019-12-25:
**Describe the bug**
I am trying to load my trezor with a recovery seed, and I want to only use the first 4 letters of each recovery word using the trezorctl python program. I have tried a few different orderings of the  command options, and have been looking at the python code and the documentation for the `click` command line parsing library, but everything looks right.

**Firmware version or revision**
Describe firmware version or revision you are using when encountering the bug.
Using trezorctl: 0.11.5
Trezor Firmware (latest, I cant find a trezorctl command to get exact version right now)

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Wallet, Trezor Beta Wallet, Trezor Suite, Electrum]
Command line Python
 - OS: [e.g. Windows, macOS, Linux, Android]
Ubuntu 18.04
 - Browser [e.g. chrome, firefox, safari]
 - Version [e.g. 42]

**To Reproduce**
Steps to reproduce the behavior:
trezorctl recovery-device -w 24 -e
trezorctl recovery-device -w 24 --expand
trezorctl recovery-device --expand -w 24
trezorctl recovery-device -e -w 24

for example, my first word might be obtain

Please confirm action on your Trezor device
Enter one word of mnemonic: obta 
trezorlib.exceptions.TrezorFailure: DataError: Word not found in a wordlist

**Expected behavior**
A clear and concise description of what you expected to happen.
According to the code and the docs, if you pass --expand, if the 4 letters isn't in the wordlist, it searches for words that start with those 4 letters.

ui.py:
`
    def expand_word(word):
        if not expand:
            return word
        if word in wordlist:
            return word
        matches = [w for w in wordlist if w.startswith(word)]
`

**Background**
I am storing my words as a SLIP39 shamir, and because each share gets quite large the more characters in the secret, I am looking to only store the first 4 characters. (I don't know how the other trezorctl command manages to fit each share into 33 words, maybe I should check out that code too.)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

prusnak, 2019-12-25:
@sako0938 which firmware version is that?

prusnak, 2019-12-25:
Which Trezor model is that?

There is something odd with the report.
The `--expand` option is for expanding words on the host (computer) and this works only for simple mnemonics (not Shamir SLIP39).
For Model T you can't enter words on the host (computer), you need to enter them directly on the device.

sako0938, 2019-12-25:
I am using a Trezor One.

I am restoring a seed by typing them on a computer, but I am looking to type in the first 4 characters not the whole word.

Sorry I brought into the equation at all about Shamirs and SLIP39, it isnt relevant here really, that is something I am doing on my own.

Still I have an issue restoring a seed, with the expand option. Please, do you know why the expand might be working, or is there a better forum to ask this question? Closing my issue right away just because I explained poorly, not sure where to go next.

prusnak, 2019-12-26:
Ok, so it seems that @sako0938 is having problems with standard 24 word recovery.  @matejcik can you have a look?

matejcik, 2019-12-27:
confirmed and fixed, thanks for the report

closed 2019-12-27 by matejcik

matejcik, 2019-12-27:
fwiw, @sako0938 :

> I am storing my words as a SLIP39 shamir, and because each share gets quite large the more characters in the secret, I am looking to only store the first 4 characters. (I don't know how the other trezorctl command manages to fit each share into 33 words, maybe I should check out that code too.)

Storing the literal words, even if only first 4 characters, is going to be hugely inefficient: each word is chosen from a list of 2048, so it only carries 11 bits of information. But to store the 4-character prefix, you are consuming 4 bytes = 32 bits, so almost 3 times as much.

You should instead convert each word into an index in the wordlist, encode it as a 11-bit number, and pack the results after one another to obtain 11 * 24 = 264-bit value, which will encode as 34-word Shamir shares if i'm not mistaken. You can even strip the final 8-bit checksum, and recalculate it when you recover the Shamir backup. This gets you down to 256 bits / 33 words.
The checksum is protecting the mnemonic _string_, which you will be reconstructing anyway, and SLIP39 has its own checksums protecting it.

sako0938, 2019-12-27:
Awesome! Thank you so much for the fix, and the advice on shamirs. That will help a lot!

commit 700c5e4f (2019-12-27): python: fix expand-words functionality in recovery (fixes #778)