trezor/trezor-firmware #788: python-trezor: Build information outdated

brandsimon, 2020-01-06:
I wanted to install python-trezor from the git repository.
In the Docs there is mentioned to run `python3 setup.py prebuild`, which says: `error: Deprecated. Run 'make gen' instead.`
Also the python docs point to the old repository: `https://github.com/trezor/python-trezor`.
Running `make gen` results in the error, that src/trezorlib/coins.json got removed, but not recreated.
Can you please update the build information?

matejcik, 2020-01-06:
the `make gen` problem is also related to #752

matejcik, 2020-01-06:
FWIW: you don't need to run `make gen` at all actually. use git to restore the `coins.json` file and proceed to install with `python3 setup.py develop`

brandsimon, 2020-01-06:
@matejcik 
Yes, this worked for me, thank you

commit f506bd44 (2020-01-24): python: update README (fixes #788)
closed 2020-01-24 by matejcik