trezor/trezor-firmware #790: FIDO2 Register screen freezes with no error message

mmahut, 2020-01-09:
**Describe the bug**

When creating a new credentials the party aborts the creation process, Trezor stays in the _FIDO2 Register_ screen without an error or a way to exit.

**Firmware version or revision**
```
Features (102 bytes) {
    device_id: '8FD6A92C73B51F1403A27057',
    flags: 0,
    initialized: True,
    label: 'My Trezor',
    language: 'en-US',
    major_version: 2,
    minor_version: 2,
    model: 'T',
    needs_backup: True,
    no_backup: False,
    passphrase_cached: False,
    passphrase_protection: False,
    patch_version: 0,
    pin_cached: True,
    pin_protection: True,
    recovery_mode: False,
    revision: 8 bytes b'02724da0',
    unfinished_backup: False,
    vendor: 'trezor.io',
}
```

**Desktop/smartphone setup (please complete the following information):**
 - OS: NixOS unstable + ```libfido2-1.3.0```

**To Reproduce**
Steps to reproduce the behavior:

1. Let's create a new credential
```
echo credential challenge | openssl sha256 -binary | base64 > cred_param
echo relying party >> cred_param
echo user name >> cred_param
dd if=/dev/urandom bs=1 count=32 | base64 >> cred_param
fido2-cred -M -h -i cred_param /dev/hidraw6 | fido2-cred -V -h -o cred
```
2. _FIDO2 Register_ screen appears
3. Abort fido2-cred
4. The _FIDO2 Register_ screen is now frozen and cannot be exited

**Expected behavior**
Error the screen if it is possible or at least permit the user to exit it.

andrewkozlik, 2020-01-09:
I tried to reproduce this on the current master build. Unable to reproduce on the emulator. On hardware, the red X button works correctly even when fido2-cred is aborted, but pressing the green button causes the screen to "freeze". It's a soft freeze, only until the next command comes along. Will investigate further.

andrewkozlik, 2020-01-09:
So the difference between the confirm and decline message is that decline fits into one USB HID packet, but confirm needs to be sent in multiple packets. When Trezor freezes, it's because it's waiting for a confirmation on the first packet before it can send the next packet. Fixed in https://github.com/trezor/trezor-firmware/pull/791 by checking for timeout.

prusnak, 2020-01-11:
Fixed and closed via https://github.com/trezor/trezor-firmware/pull/791

Thanks for the report and for the fix!

closed 2020-01-11 by prusnak