trezor/trezor-firmware #792: python trezor - unexpected keyword

noahsilverman, 2020-01-12:
Installed most recent version of python trezor using 
`pip install trezor`  

System:
OSX  10.14.6
Anaconda Python 3.7.4

Version installed: trezor-0.11.6

Calling: trezorctl from command line produces and error:

`  
Traceback (most recent call last):
  File "/Users/noah/anaconda3/bin/trezorctl", line 6, in <module>
    from trezorlib.cli.trezorctl import cli
  File "/Users/noah/anaconda3/lib/python3.7/site-packages/trezorlib/cli/trezorctl.py", line 28, in <module>
    from . import (
  File "/Users/noah/anaconda3/lib/python3.7/site-packages/trezorlib/cli/btc.py", line 104, in <module>
    @click.pass_obj
  File "/Users/noah/anaconda3/lib/python3.7/site-packages/click/decorators.py", line 171, in decorator
    _param_memo(f, OptionClass(param_decls, **attrs))
  File "/Users/noah/anaconda3/lib/python3.7/site-packages/click/core.py", line 1454, in __init__
    Parameter.__init__(self, param_decls, type=type, **attrs)
TypeError: __init__() got an unexpected keyword argument 'hidden'
`

Looking into the code, I see that line 102 of btc.py sets the arguments as:
`@click.option("-c", "--coin", "_ignore", is_flag=True, hidden=True, expose_value=False)`  

However, core.py Options class, which starts on line 1414 does not have a parameter defined for "hidden".  Thus, causing the crash.

noahsilverman, 2020-01-12:
Note:  Installing the trezorlib directly from github source works fine.  So, perhaps the issue is related to how Anaconda pip is doing the install.

matejcik, 2020-01-14:
please find out what your click version is.
in click 7.x, which is [required](https://github.com/trezor/trezor-firmware/blob/master/python/setup.py#L13), the option exists

matejcik, 2020-01-21:
closing this because I'm pretty sure that Anaconda is not honoring trezorlib's requirement for Click 7.X
@noahsilverman please let us know if there is more to this.

closed 2020-01-21 by matejcik