trezor/trezor-firmware #793: Can't build Legacy emulator on NixOS

tsusanka, 2020-01-15:
**Describe the bug**
Can't build Legacy emulator.

**Firmware version or revision**
Current master - 783fa80d19c1312d0c6fbbd43ad7bcbc38607db1

**To Reproduce**
- nix-shell
- cd legacy
- export EMULATOR=1 DEBUG_LINK=1 
- script/setup
- script/cibuild

**Error message**
```
make: Entering directory '/home/tsusanka/Projects/trezor/firmware/legacy/emulator'
MEMORY_PROTECT=1
  CC      setup.o
setup.c:27:10: fatal error: libopencm3/stm32/flash.h: No such file or directory
   27 | #include <libopencm3/stm32/flash.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

-----

Since this works on CI and @matejcik's Mint, I suppose this is somehow related to NixOS. cc @mmahut @1000101 just in case they have an idea..

tsusanka, 2020-01-15:
I'm not sure this actually helps, but this command fails the very same way:

```
gcc -I/home/tsusanka/Projects/trezor/firmware/legacy/vendor/libopencm3/include -c setup.c
```

matejcik, 2020-01-15:
> ```
> gcc -I/.../legacy/vendor/libopencm3/include -c setup.c
> ```

is this a typo? the `-I` definitely shouldn't start with `/.../`

tsusanka, 2020-01-15:
That was an indication, that there is my directory. Fixed to make it clearer

prusnak, 2020-01-16:
Fixed via #794

closed 2020-01-16 by prusnak