trezor/trezor-firmware #797: Hardware wallet workshop

dbolser, 2020-01-20:
I run a bitcoin meetup in Cambridge, UK
https://meetup.com/CambridgeBitcoin/

Currently we have about 600 members.

We are thinking to organise a series of 'tutorials', one of which will be focused on hardware wallets and key management. (I currently have about 6 OpenDimes and a ColdCard ;-)

I was wondering if you could gift us a few items to give away as prizes during the tutorial?

Else do you have any other promotional material that you give away?

Out of interest, do you run any workshops?

Many thanks,
Dan.

prusnak, 2020-01-20:
Please contact us at press@satoshilabs.com

This is the issue tracker for developers.

closed 2020-01-20 by prusnak

dbolser, 2020-01-20:
Thanks, posted here out of frustration (looking for the contact).