trezor/trezor-firmware #798: Enable adding custom ERC20 tokens to my Trezor Device

SergioDemianLerner, 2020-01-20:
**Problem**

The Trezor wallet manages a small set of pre-defined tokens, for which it shows the token name and correct number of decimals based on the token definition.
Users need to transact securely in all tokens available.

**Solution**

Because any statistically linked list of predefined tokens will always be out-of-date, we suggest this list is dynamic.
The proposed solution is to create and API and a UI to load custom tokens into the device. When adding a token, the user should specify the address, network and decimal places. The device should store the descriptions in non-volatile memory. The device should require confirmation for adding or modifying the description of a token.

**Alternatives Considered**

Other alternatives are systems for automatically discovering new ERC20 tokens based on cumulative proof of work (complex and incompatible with many blockchains) or that Satoshi Labs scans all blockchains continuously for new token contracts and sends this information weekly to the devices as partial upgrades (complex, centralized, and costly for Satoshi Labs).

prusnak, 2020-01-20:
Leaving this responsibility to the user is not acceptable from the usability perspective.

Where I do see the way - we create a mechanism where it would be possible to do a partial upgrade with the signed token definition generated regularly from https://github.com/ethereum-lists/tokens (the list we already use).

SergioDemianLerner, 2020-01-20:
The problem is that the user is sometimes not a newcomer. It can be a token development team, advisor, early investors, or financial team member.  Generally they need to securely manage tokens before they are widely known and accepted in those lists. Maybe the feature could be added in the form of an API call and a simple command line interface that works well for advanced users, and no UI is shown to the end-user. Maybe I could contribute with coding such feature, if it will be accepted.

prusnak, 2020-01-21:
No, it won't be accepted. The whole point of Trezor is that it provides a trusted display. If such API was in place a malicious actor can add malicious token deceiving the user defeating the whole point of having a trusted display.

If you need something to show to your investors you are welcome to develop your own firmware and show them this.

That said, working on a solution that takes tokens definition from a curated list (such as the one above) and pushes them into a Trezor as a micro-update would be appreciated.

matejcik, 2020-01-21:
linking to #15 for reference, the same or similar mechanism could be used for tokens

Also noting that we could improve UX of unknown tokens, by (1) showing wei amount of the token being transacted, and (2) showing the contract address on the second page of the confirm dialog

SergioDemianLerner, 2020-01-21:
I think it's easier to perform a global attack to the "curated" list than use the advanced feature that I proposed.
For instance, let's support I manage to add the token "freetoken" into the "curated" list but I copy the contract address of chainlink token (and nobody notices it). Then I go telling people they have "freetokens" in their wallets which can be sent to me and I will send back some tiny amount of eth in return, which I really do.
People empty their chainlink tokens without even knowing they are doing it.

That's a risk with externally curated lists.

prusnak, 2020-01-21:
The externally curated list does not mean we are not reviewing the changes when we pull the list in.

prusnak, 2020-01-21:
Anyway, I am closing this in favour of https://github.com/trezor/trezor-firmware/issues/800 (Show Wei amount and contract address for unknown tokens) and https://github.com/trezor/trezor-firmware/issues/15 (SignTx should contain the signed definition of the used coin/token)

closed 2020-01-21 by prusnak