trezor/trezor-firmware #802: Prepare script to combine bootloader and firmware incorrect?

braydonf, 2020-01-23:
**Description** 

The script at `legacy/bootloader/combine/prepare.py` appears to not be correct.

```py
combined = bl + fw[:256] + (32768 - 256) * "\x00" + fw[256:]
```
As from `legacy/memory.h`, the expected layout is `bootloader (32768)` +  `storage (32768)` + `firmware header (1024)` + `firmware app (982016)`.

From what I can tell, it appears that this would be more correct:
```py
combined = bl + 32768 * "\xff" + fw
```
Does the storage area need to be initialized? From what I can tell from `legacy/bootloader/usb.c` the erased value is expected to be `0xff` for the entire storage size.

**Firmware version**
legacy/v1.8.3 _(and others)_

prusnak, 2020-01-23:
Good catch. The script was not updated to the new memory layout. Fixed in 0b3d0c66816c980cee64971cc70b636ba9037fb8

closed 2020-01-23 by prusnak

braydonf, 2020-01-23:
Okay cool. Looks like the last version with the previous format was `v1.7.3` at [`memory.h`](https://github.com/trezor/trezor-mcu/blob/f641e798f91a15c3b09e8dc6a163195dd56f86d2/memory.h). So this relates to `v1.8.0`, `v1.8.1` and `v1.8.3`.