trezor/trezor-firmware #806: SD protection add one LOCKED screen

bosomt, 2020-01-27:
**Describe the bug**
SD protection adds one LOCKED screen

**Firmware version or revision**
2.2.1 + 2.2.1.btc  /2.2.0 OK/

**Desktop/smartphone setup (please complete the following information):**
any computer 

**To Reproduce**
Steps to reproduce the behavior:
enable SD protection
start device
see video: https://photos.app.goo.gl/xKppd31w1eDGj98t9

**Expected behavior**
lock screen is available only when there's PIN protection

**Screenshots**
https://photos.app.goo.gl/xKppd31w1eDGj98t9

**Additional context**
when PIN code is enabled then it works fine and you only see one locked screen.

tsusanka, 2020-01-27:
Note that this might be the way we want it. Let's discuss it IRL.

andrewkozlik, 2020-01-28:
This behavior is intentional, but if it's confusing then we can easily change it. The motivation behind it was that if the SD card is not present when the device is powered on, then the first thing the user will see is an error dialog. It seems nicer to first show some kind of welcome screen. Secondly, it indicates to the user that some kind of protection is in place.

prusnak, 2020-01-28:
This is indeed not a problem. Let's keep it this way.

closed 2020-01-28 by prusnak