trezor/trezor-firmware #807: SD protect add format feature

bosomt, 2020-01-27:
**Is your feature request related to a problem? Please describe.**
Please add format feature to SD protect feature, it's confusing that there is no info about required SD card format.

**Describe the solution you'd like**
trezorctl device sd-protect format

andrewkozlik, 2020-01-28:
Good point. Trezor could also offer to format the card if it finds that the filesystem is not recognized. If the user would choose to proceed, then it would show a hold-to-confirm dialog alerting the user that all the data on the SD card will be lost.

prusnak, 2020-02-07:
Fix in https://github.com/trezor/trezor-firmware/pull/833

prusnak, 2020-02-07:
I will soon push FatFS update to master which includes:
* `FatFS.mkfs()` for the device (previously it was enabled only for the emulator)
* `FatFS.setlabel()`

We'd need to adapt the `ensure_filesystem` decorator so it does the following instead of `fs.mount()`:

``` python
try:
    fs = io.FatFS()
    fs.mount()
except OSError:
    text = Text("SD card error", ui.ICON_WRONG, ui.RED)
    text.normal("Unrecognized filesystem")
    text.normal("found. Unplug the device")
    text.normal("and remove the card or")
    text.normal("you can format the card")
    text.normal("losing all the data on it.")
    dialog = HoldToConfirm(text, confirm="Format SD card")
    await dialog
    fs.mkfs()
    fs.setlabel("TREZOR")
    fs.mount()
```

This will not be that easy, because the decorator is not a coroutine and we cannot call `await` inside. That's why I am leaving the final fix to someone else who understands async better than me.

prusnak, 2020-02-07:
More UI suggestions by andrew here: https://github.com/trezor/trezor-firmware/pull/833#discussion_r376432774

matejcik, 2020-02-20:
QA scenarios:

* use a FAT32-formatted SD card, try Trezor's SD features, they should work as usual
* use an unformatted SD card, try Trezor's SD features. Trezor should ask whether you wish to format. If you format the card, Trezor should work fine, and the SD card should be readable on PC
* if you don't format the card, Trezor should cleanly inform you of the problem

matejcik, 2020-02-20:
a quirk of the current setup: if you have SD protection enabled, and insert an unformatted SD card, Trezor will first ask to format the card -- even though it could technically just reject with "wrong SD card" immediately.

not sure if we want to change this?

andrewkozlik, 2020-02-20:
> a quirk of the current setup: if you have SD protection enabled, and insert an unformatted SD card, Trezor will first ask to format the card -- even though it could technically just reject with "wrong SD card" immediately.
> 
> not sure if we want to change this?

Good point. I think we should reject the card in that case.

closed 2020-03-20 by tsusanka

bosomt, 2020-07-20:
**QA OK**

solved via https://github.com/trezor/trezor-firmware/pull/833

PR #858 (2020-02-20): Format SD card when appropriate
fixes #807 

This does not pass tests yet, one known problem is that the "overwrite SD card with empty bytes" feature is broken. However, the application level code can be reviewed and the formatting feature seems to work in practical testing.

I didn't try on an actual device yet.