trezor/trezor-firmware #809: Bitcoin-only firmware responds incorrectly to non-Bitcoin messages

tsusanka, 2020-01-27:
**Describe the bug**
When Bitcoin-only firmware receives an altcoin message it should respond with an Unexpected message error.

However, the emulator fails on:
```
Traceback (most recent call last):
  File "trezor/wire/__init__.py", line 289, in handle_session
  File "trezor/messages/__init__.py", line 30, in get_type
KeyError: 307
```

**Firmware version or revision**
master  cf73d8e499f73d0293c9a5b3580c3c62e0323e9b

**To Reproduce**
Start emulator.
Run  `trezorctl cardano get-address -n "44'/1815'/0'/0/0"` 

-----

@trezor/qa could you retest this on device with the latest v2.2.0?

bosomt, 2020-01-27:
2.2.0.btc:

bosomt, 2020-01-27:
same results with 2.2.1.btc

tsusanka, 2020-01-27:
Thanks. It's only emulator-related then as I suspected.

It's a tiny bug but it has the ugly consequence that the test suite for bitcoin only firmware will hang if you forget to mark it using `@pytest.mark.altcoin`.

---

We still should fix this but - @matejcik how about we add a timeout for each test function? Then we would see on CI that this test failed on timeout, which is significantly less confusing than hanging test suite. Should not be a problem on the pytest side, see e.g. https://pypi.org/project/pytest-timeout/.

matejcik, 2020-01-27:
AFAICT it's a matter of adding the plugin to the pipenv and setting some reasonable default in the config file, so there's literally no downside :)

matejcik, 2020-01-27:
timeouts configured in 88ee780125bfdc518a716e751e8f7ebcfba5816e and 066686a20c640a2dd449ce89a3204c0ef10aa579, let's see what happens in CI

commit 0df3c648 (2020-01-27): core: fix unexpected messages in debug mode (fixes #809)
closed 2020-01-27 by matejcik