trezor/trezor-firmware #815: Dry run for supershamir with one share asks for one more after reconnect

bosomt, 2020-01-31:
**Describe the bug**
Dry run for supershamir with one share asks for one more after reconnect when asked to press continue.
Recovery works fine.

**Firmware version or revision**
2.2.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: beta
 - OS: win 10
 - Browser chrome
 - Version latest

**To Reproduce**
Steps to reproduce the behavior:
start dry run
enter 1 of 1 share from group
don't press continue but reconnect device
you are asked to enter share

**Expected behavior**
dry run should end /i think/

**Screenshots**
[image]

bosomt, 2020-01-31:
seed: https://pastebin.com/MHHLBjjZ

tsusanka, 2020-01-31:
@bosomt please retest with https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/jobs/422067169/artifacts/download

bosomt, 2020-02-03:
@tsusanka works fine!

tsusanka, 2020-02-03:
Closed by https://github.com/trezor/trezor-firmware/commit/277e494ffa42fbd3ae4e747e5f0e60c99251070e

closed 2020-02-03 by tsusanka

bosomt, 2020-07-20:
**QA OK**

FW 2.3.2 e70811c
wallet b3f18eb73765f92dd35eff2c562ef1edb5dd9fa3
latets nixos and Chrome