trezor/trezor-firmware #817: Numeric speed recovery

dolebas, 2020-02-02:
**Problem**

I am not the only one who wipes my device after using it, for protection agains physical attacks. The recovery process is already quite convenient but still takes some time.

**Solution**

For advanced users, the recovery process would be quicker with a numeric keypad, for alternatively entering the row numbers in the BIP39 wordlist, instead of the words.

Switching between regular input mode and numeric keypad could work the same way as switching between numeric/alpha when entering a passphrase, by swiping.

For avoiding typos, the matching BIP39 word could show up for confirmation after intervals of 4 digits.

**Alternatives**

A passphrase can also protect agains physical attacks but it only allows 50 characters. 12 words represented as 4 digits each can fit in a passphrase but it is not an optimal solution because it is easy to make a typo. It also requires two independently generated seeds and backups, which can cause confusion.

A simple one-word passphrase provides no significant protection agains physical attacks.

tsusanka, 2020-02-02:
Did you try out SLIP-39 aka Shamir Backup? For Shamir Backup we designed a special keyboard where each word is uniquely defined by pressing at most four different buttons. That means there is no delay when you have "similar" letters (letters on the same button) because there are no such words.

dolebas, 2020-02-02:
Yes, I like the idea with the special keyboard for Shamir Backup but haven't been able to speed up the process with 20 word Shamir, compared to a regular 12 word mnemonic.

The regular layout is still familiar from the pre-smartphone era, when SMS were typed that way. It probably takes some time for the brain to adjust.

12 words on the regular layout takes about 90 sec for me but when I simulate numeric recovery with 48 integers using the passphrase feat it only takes 30 sec.

tsusanka, 2020-08-04:
Hi there, thank you for your suggestion. After a thorough discussion in our Product team we have decided we do not wish to implement this feature. We need to prioritize our backlog and focus on features that bring the most value to our customers. We also need to restrict the feature set to keep the firmware small and maintainable. Thank you for your understanding.

closed 2020-08-04 by tsusanka