trezor/trezor-firmware #820: Fix flashing bootloader from SD card in boardloader after chaning SD card to use DMA

hiviah, 2020-02-03:
Flashing bootloader from boardloader doesn't work anymore after switching to use of DMA for SD card transfers.

When attempting to flash bootloader from SD card now, the code gets stuck in infinite cycle in `SystemInit` from `embed/trezorhal/stm32.c`.

hiviah, 2020-02-11:
Usign normal STLink while debugging boardloader makes it stuck upon system reset in `SystemInit` when oscillators switch from HSI to HSE. Black Magic Probe does not seem to suffer from this issue.

There are 2 interrupt handlers to go with SD card, `DMA2_Stream3_IRQHandler` and `SDIO_IRQHandler` which somewhere in depths call `HAL_SD_IRQHandler`.

In functioning case (e.g. firmware) there should be the same number of DMA and SDIO IRQs (can be seen by enabling irq_stats field).

Strange thing is that just by breaking on DMA handler in debugger, the SDIO handler starts coming. It's always 5 SDIO interrupts and one DMA interrupt no matter how many delay/timeouts or how much time you wait in the debugger

hiviah, 2020-02-21:
Apparenty STM32F4xx can't do read/write via DMA to CCMRAM (0x10000000), only to SRAM (0x20000000). Which results in the strange behavior only in boardloader which is put in CCMRAM and not in firmware.

hiviah, 2020-02-24:
Either of moving boardloader to SRAM from CCMRAM in linker script or using buffer at some absolute SRAM address seems to fix this.

prusnak, 2020-02-25:
Fix in https://github.com/trezor/trezor-firmware/pull/876

closed 2020-02-27 by prusnak

tsusanka, 2020-03-20:
@prusnak @hiviah do I understand correctly that this issue is "QA-covered" by the others and I can apply the `doesn't need QA` tag?

prusnak, 2020-03-20:
@tsusanka right, I just did apply the tag